pvaret/rtl8192cu-fixes #28: You are a godsend.

flinz, 2015-01-11:
Thanks so much for this!

I had the problem of my Edimax EW-7612UAn v2 dropping connections after a few minutes with the legacy 14.04 RTL8192cu drivers.

I fiddled around with trying to compile the drivers provided by RTL and edimax (both the rtl8712_8188_8191_8192SU_usb_linux_v2.6.6.0.20120405 drivers), to no avail.

Yours worked out of the box and I am now connected for an hour without connection dropping.

Thanks again!

pvaret, 2015-01-11:
Thank you for your kind message. Truly appreciated. :)

closed 2015-01-11 by pvaret

avimar, 2015-02-03:
Exact same story here. But it was really hard to finally find your repo... so I've spread links back to all the places I originally was.

Thanks!!