pvaret/rtl8192cu-fixes #29: IS it possibile to make a patch for the kernel tree?

threarth, 2015-01-31:
Hi,
I've just built your module for a TP-LINK TL-WN821 N which should have mounted an atheros chipset (google search results) but when I plugged in i discovered that instead in there was a realtek rtl8192cu chipset which worked very poorly (unstable connection or no connection at all with many loss packets).
I built your driver and this device is running fine, i think that these changes should be included in the default kernel tree, may you do it?
I think i could prepare a patch if you agree,

regards

dlech, 2015-01-31:
The place to ask this question is the linux-wireless mailing list. https://wireless.wiki.kernel.org/en/developers/mailinglists

I've never asked, but I kind of assume that if this driver was fit for inclusion in the mainline kernel that someone would have done it already.

But, I suppose it doesn't hurt to ask.

pvaret, 2015-02-02:
Hi threarth, hi dlech,

I've actually been owing Greg Groah-Hartman a patch for months. :) I need to get around to it!

snobel, 2015-02-19:
Thanks for providing this! Made my TL-WN823N work with my Netgear WNR3500L router, which wasn't even visible before. 

Also thanks for the easy to follow build instructions - still, having these fixes mainlined would be great...

Pastafarianist, 2015-04-27:
I had major problems with my TP-LINK TL-WN823N (really slow and unstable connection, about 20% lost packets, complete failure after suspend & restore), but thanks to your module (and a friend who put together an [ebuild for Gentoo](https://gitweb.gentoo.org/dev/maksbotan.git/tree/sys-kernel/rtl8192cu-fixes/rtl8192cu-fixes-9999.ebuild)) these problems are now gone and everything is running fine. Someone should really push to include this in the kernel!

pvaret, 2015-04-27:
Thank you for the feedback! I added the link to that ebuild in the README.

As to getting this into the kernel, honestly, it's my fault: I've been meaning to submit it for ages, and just never got around to it. Meanwhile I leave this ticket open as a reminder.

dlech, 2015-04-28:
Raspberry Pi has had this in their kernel for a few years. You should be able to copy their patches and make quick work of it.

https://github.com/raspberrypi/linux/commits/rpi-4.0.y/drivers/net/wireless/rtl8192cu

pvaret, 2015-04-28:
Thank you the pointer, dlech. The code you linked to appears to use the `*_proc_entry` functions that are deprecated in mainline kernels, and required me to start this project in the first place. I don't know how it even compiles in the first place. Did the RPi people un-deprecate these functions?

dlech, 2015-04-28:
All of the calls to `*_proc_entry` are inside of a `#ifdef CONFIG_PROC_DEBUG`, so they are not compiled.

dlech, 2015-04-28:
FWIF, I did a diff from the RPi patch to this repo and filtering out whitespace changes, there are only a couple differences. Basically, it is missing 312963f8e83814e2d7492966aa691026c7057afc and 0c0350fb5a180c67b7b91e07d7a18fa43c5721ab and thats it. I cherry-picked the RPi commits and they applied cleanly to a (mostly) mainline 3.16.7 kernel.

pvaret, 2015-05-03:
Ok, this is good to know, thanks. I still think I prefer the approach in this repository better -- explicitly checking against the kernel version -- but at any rate, it's good to know that there is no technical holdup to getting this into the kernel.

arehtykitna, 2015-05-06:
I prefer your kernel specific approach too. Good luck submitting the patch

tytower, 2015-05-11:
Jeez I've had some problems trying to get this fixed in the kernel by someone but didn't know where to go . I too would appreciate a kermel fix .Hope its going well

pvaret, 2015-05-12:
I've got a few days off coming up and will try to see it done then. Phew.

arehtykitna, 2015-05-12:
Thank you very much, please don't work too hard on your days off though.

fat-lobyte, 2015-05-28:
There is a dedicated page on how to submit patches to the Wireless subsystem here:
https://wireless.wiki.kernel.org/en/developers/documentation/submittingpatches

If you need assistance, please comment on this bug. I would really love to see your stuff merged into mainline. I am running it with DKMS for now (also a big thanks for providing a dkms.conf!).

Pastafarianist, 2015-07-06:
Has there been any progress on submitting the patch?

pvaret, 2015-07-06:
I've barely been home lately, so I'm sorry to say there hasn't been. I'm keenly aware of the number of pending tickets in the system that are still waiting for my input, believe me. :/

pvaret, 2015-07-06:
@fat-lobyte: Thanks for the link. That page is about sending in up to medium-sized patches, by email, though, and I don't know if an entire new driver would fit the bill. Worth a try, I guess?

pvaret, 2015-07-06:
All RIGHT. Due to exceptional circumstances, I had a lot of free time today at work. End result: http://driverdev.linuxdriverproject.org/pipermail/driverdev-devel/2015-July/072377.html

Hopefully this will suffice to get things in motion.

arehtykitna, 2015-07-06:
Thank you for all your efforts! Hopefully it gets the ball rolling. On another note, your drivers are still working fine after upgrading my Father's PC to LM 17.2 and kernels from 3.13 to 3.19 series. No re-install of your patch was needed, the update process took care of that too.
Cheers!

pvaret, 2015-07-07:
Yes, that's the good thing about DKMS: the driver is rebuilt automagically on kernel upgrades... unless the kernel API changed in the meanwhile. Ideally the best outcome would still be if the kernel maintainers fixed the original in-kernel driver. But until then...

pvaret, 2015-07-07:
All right, the Linux kernel driver team is not too keen on merging this driver. I can understand, really: there's already an in-kernel driver for these devices, and it would be better to just fix it, rather than merge a pretty big and messy lump of code that reimplements everything instead of using the kernel's own wireless stack. We'll see what happens now.

tytower, 2015-07-08:
OK so who can I pester to try to move it on a bit please . Do you have a contact?

-------------------------------------------------------------------------
On Tue, 07 Jul 2015 14:43:38 -0700
"P.

pvaret, 2015-07-08:
I linked the thread in an earlier comment, but please don't go and pester anyone. Expressing your need is okay, but allow me to remind you that most of the people involved are unpaid volunteers doing what they do on their own free time.

tytower, 2015-07-08:
A lesson in life you have obviously not placed enough emphasis on . The squeaky wheel gets the oil.

----------------------------------------------------------------------On Wed, 08 Jul 2015 05:55:25 -0700
"P.

fat-lobyte, 2015-07-16:
> All RIGHT. Due to exceptional circumstances, I had a lot of free time today at work. End result: http://driverdev.linuxdriverproject.org/pipermail/driverdev-devel/2015-July/072377.html
>
>Hopefully this will suffice to get things in motion.
Oh man, that thread is a frustrating read. Thanks for your efforts pvaret!
So what's the state now? Are you going to try to fix the existing driver? Or are you trying again on the linux-wireless mailing list?

You mentioned
> It also lacks a proper port of the procfs populating code to the new procfs API, 
though that doesn't prevent it from working. (I only disabled the code that uses 
the old API on kernel versions > 3.9.)

Do you think that if this were implemented, that the acceptance to merge this driver would be higher?

tytower, 2015-07-16:
The problem as I see it is the old non functioning driver is going out with every new linux kernel release and so, to use the new kernel, a large portion of the linux community must find someone in their distribution followers  to recompile the manufacturer's driver for that kernel or else they have to stay with the old . These wireless drivers are important as otherwise you have a flaming long curly wire running through the house that every bastard trips over and pulls something off the bench .
I think the "pestering " is not getting to the right people in the kernel updating operation and perhaps we should pester a few more in a different place . Can someone else try to get them moving on this .https://wireless.wiki.kernel.org/en/developers/documentation/submittingpatches

pvaret, 2015-07-19:
>Do you think that if this were implemented, that the acceptance to merge this driver would be higher?

I doubt it. The problem as I understand it is that the in-kernel drivers share a lot of the common infrastructure and only implement the differences between each devices, where this driver shares nothing with anyone -- it implements everything on its own. That makes it huge, and therefore difficult to review, and difficult to maintain. The kernel developers feel it's not worth it and the proper solution would be to just fix the existing in-kernel driver instead.

Anyway. The patch has been refused, so I'm closing this issue.

@tytower: Again, kindly don't pester anyone. It's done. This driver is just not getting into the kernel. If you want to do something useful, you should consider offering your help fixing the in-kernel driver. There's an open bug report for it, that should be a good starting point.

closed 2015-07-19 by pvaret

tytower, 2015-07-20:
This is silly . One minute you are fixing it , then you are not , then you are fixing it again , then you are not . 

Recompiling the factory driver with the new kernel fixes the problem so how damn hard can it be to fix it properly or give it a bloody new name . 

You are just glory seekers with an oppressive , predetermined outcome that does not get any-thing fixed and I have been following your footsteps for 3 years or more and have watched this joke site. 
Bloody ridiculous.

-----------------------------------------------------------------------------------On Sun, 19 Jul 2015 15:01:01 -0700
"P.

tytower, 2015-07-20:
And by the way don't puport to tell me what to do Moron
-------------------------------------------------------------------------------On Sun, 19 Jul 2015 15:01:01 -0700
"P.

pvaret, 2015-07-27:
Thank you kindly for your helpful feedback.

tytower, 2015-07-29:
Yes always trying with the smart comments but not getting the job done though. Don't close it until it is fixed . Out of sight out of mind does not hold with me . Do the work if you want the Glory

arehtykitna, 2015-07-29:
@pvaret - thanks for trying.

IonicaBizau, 2015-09-12:
@dlech Can you take a look [at this question](http://raspberrypi.stackexchange.com/q/34278/33613)? Maybe you can help me. `lsusb` lists my wifi adapter, but `iwconfig` doesn't. I guess it's a driver issue (didn't installed any drivers for it *yet*).