pvaret/rtl8192cu-fixes #32: Ad-Hoc mode

FVerbum, 2015-02-15:
Hello.
I use this driver for USB dongle EDUP EP-MS1537 (8192CU) on Ubuntu 14.04 (3.13.0-45-generic).
Unfortunately I can't setup Ad-Hoc mode (static ip without wpa) - network manager say that all ok - connect on,
but I can't ping any comp in net.
I tried without network manager (unmanaged-devices and settings in .../interfaces) same result.
In same time internal wifi adapter Intel 2200 works without problem.
And this dongle on RaspberryPi (Raspbian 3.12.35+) in Ad-Hoc mode works without problem.
Can you help me?

pvaret, 2015-02-16:
Wish I could. It seems to be pretty random as to what works on what hardware. :/

Does the dongle work in infrastructure mode at least?

FVerbum, 2015-02-16:
Yes, dongle works in infrastructure mode no problem on 300mb/s.

pvaret, 2015-02-16:
It *may* be that ad-hoc just isn't implemented (or isn't implemented properly) in the driver. As I've expounded at length elsewhere, this here is the default Realtek driver, just banged on until it would compile against recent kernels. Let the records show that the default Realtek driver is *not* the best written piece of software ever to be thrown at a computer anywhere.

What I'm trying to say here is, I'll leave this ticket open for now just in case something turns up, but don't hold your breath.

dlech, 2016-05-05:
I've been doing some research on this. The best answer is <https://github.com/lostincynicism/hostapd-rtl8188>. TL;DR; It is not going to work unless you use hostapd with patches from Realtek.

There are also some recompiled binaries for RPi at <https://github.com/lostincynicism/hostapd-rtl8188>.

dlech, 2016-05-09:
After further research, it looks like <https://github.com/lostincynicism/hostapd-rtl8188> does not work for some chipsets (for example `0bda:8176` is the one I am using). They only thing I was able to get to work is <https://github.com/jenssegers/RTL8188-hostapd>. The nice thing about it is that it is easy to download, compile and install.

closed 2017-10-14 by pvaret