pvaret/rtl8192cu-fixes #33: Does not work with latest stable kernel [v3.19.3]

akossomfai, 2015-04-07:
The module compiles and loads correctly against the latest stable kernel [v3.19.3], does not show any problems in dmesg; but the device does not show up among the wireless interfaces. Tried on earlier kernel versions (v3.16.0, v3.16.1) and there it works correctly with no errors.
Tested all cases on Ubuntu 14.04 LTS with an Asus USB-N10 Nano.

pvaret, 2015-04-12:
Thanks for the heads up.

Does the driver show up if you run 'lsusb -t'?

akossomfai, 2015-04-13:
yes, it shows up:

somi@somi-desktop:~$ lsusb -t
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/2p, 12M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/3p, 12M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/3p, 12M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/3p, 12M
    |__ Port 3: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-pci/3p, 12M
    |__ Port 2: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/6p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/6p, 480M
    |__ Port 4: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl8192cu, 480M
somi@somi-desktop:~$ iwconfig 
eth0      no wireless extensions.

lo        no wireless extensions.

wlan1     no wireless extensions.

somi@somi-desktop:~$ lsmod | grep 8192
8192cu                491202  0

pvaret, 2015-04-14:
Thank you for the update. This is highly bothersome and I don't know where to start looking.

I'll be away for the week, but in the meanwhile, may I ask you to check what happens if you remove the file that blacklists the regular kernel driver (/etc/module.d/blacklist-native-rtl8192.conf)? Basically I'm wondering if the problem also affects the regular driver, or only this one.

akossomfai, 2015-04-19:
Tried with latest kernel 4.0 and it seems that all problems gone, it works perfectly.
However, compiling now with the latest kernel headers there is a compilation error -- just created a pull request for that.

pvaret, 2015-04-22:
Thanks. I'll need to run more tests across different versions. Noooot looking forward to that.

Taking a look at your PR right now. In the meanwhile, I'll go out on a limb and close this issue. Poke me if needed and I'll reopen it. Thank you for your help.

closed 2015-04-22 by pvaret