pvaret/rtl8192cu-fixes #35: When ported to kernel 3.16?

tohyob, 2015-04-25:
I have installed lubuntu 14.10, with kernel 3.16.
Compilation failed starting from line 313 of https://github.com/pvaret/rtl8192cu-fixes/blob/master/os_dep/linux/os_intfs.c because of function create_proc_entry deprecated.
There are other lines with create_proc_read_entry, also deprecated.
Should be replaced with proc_create and proc_create_data? I am trying to patch it, but if anyone had a working solution before me...

pvaret, 2015-04-25:
These functions were obsoleted in kernel 3.10 and are specifically #defined out in the file you refer to. I have no idea why the compilation is failing for you. Can you check you got the right file? I.e., can you see the line `#if(LINUX_VERSION_CODE < KERNEL_VERSION(3,10,0))` on line 282 of `os_intfs.c`? If not, make sure your checkout of this repo is up to date.

tohyob, 2015-04-26:
Please disregard me.... I thought I have to install first the drivers downloaded by realtek site and then fix them with rtl8192cu-fixes.... Instead the fixes must be installed "alone" following your guide.... They work like a charm.... Thank you very much!!

closed 2015-04-26 by tohyob

pvaret, 2015-04-27:
No problem! Happy to hear that your problem is fixed! :)