pvaret/rtl8192cu-fixes #38: Changing MAC ADDRESS

kheiro213, 2015-05-06:
Hi everyone,
I'm using Linux MINT 17.1 64 Bits (Ubuntu 14.04.01 based OS) and recently, I bought a TP-Link TL-WN822N v3.0 using a Realtek 8192 chipset, I needed to install the 8192cu Driver, and I followed your instructions on this link and it worked fine, but my problem is that I can't change the mac address of my WiFi Card neither by using macchanger or by manually typing:

sudo ifconfig wlan0 down
sudo ifconfig wlan0 hw ether xxxxxxx
sudo ifconfig wlan0 up

I've also tried to edit the connection by adding a cloned mac address but it wouldn't work, I'm really stuck with this, I hope that someone will help me to get it work.
Best regards,

pvaret, 2015-05-12:
Hi! Thank you for this report. I don't know if every WiFi hardware supports changing the MAC. I looked into the issue a little and couldn't find confirmation either way, so don't hold your breath.

kheiro213, 2015-05-13:
Thak's for your reply,
I could change the MAC Address before installing this wifi driver(and I can do it with every wifi card under Linux), but when I install and reboot, i can't do it anymore, should I give up ?

pvaret, 2015-05-20:
I very much doubt I'll have time to look into this in any depth anytime soon, so as said before, don't hold your breath.

That said, there are other drivers that claim to support the 8192cu chipset (like this one: https://github.com/lwfinger/rtlwifi_new). I don't know if they also support setting the MAC. Maybe it's worth a try?

alexvazquezfente, 2016-01-14:
Same issue here. My wifi chipset is detected as:
```
Bus 005 Device 002: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
```
The hardware supports changing the MAC because i can change it if i use the buggy driver rtl8192cu

@pvaret I think the alternative rtlwifi_new won't work for my hardware, given none of the drivers included there supports 0bda:8176. Do you plan to look into this issue anytime soon?

pvaret, 2016-01-14:
Somewhat unlikely, for a whole bunch of reasons. But the main of these raisons is a positive one: kernel 4.4 has been released with a brand new driver for this range of devices. Hopefully this repo here will soon be completely obsolete, and we can all use a working in-kernel driver.

alexvazquezfente, 2016-01-15:
Ok, although I will have to find alternative solutions. I cannot update kernel since i'm on an embedded system and new kernels are still unsupported. Thank you anyway.

closed 2017-10-14 by pvaret