pvaret/rtl8192cu-fixes #41: Driver does not load

concifederico, 2015-05-21:
Hi, On Ubuntu 14.04 if I blacklist the native drivers I just do not see the wireless networks. I'm newbie at driver levels. Please help!

concifederico, 2015-05-21:
Hi, sorry it DOES load OK, but my IC is RTL8188CUS, I don't know why so many forums list this as a solution for RTL8188CUS.

concifederico, 2015-05-21:
Hi again, the original driver was rtl8192cu, so 8192cu should work. Driver 8192cu is listed in lsmod, and also loads with sudo modprobe 8192cu after sudo modprobe -r rtl8192cu. After connecting the device, led does not tun on and there are no wireless networks listed. Kernel is 3.13.0-32-generic
. Device is TP-Link Nano TL-725N.

pvaret, 2015-05-21:
Hi! What's the USB id of your device? (Type 'lsusb' in a terminal, and 
there should be a line mentioning it.)

concifederico, 2015-05-22:
Thanks for your reply! Here we go:
Bus 003 Device 002: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter

pvaret, 2015-05-22:
Okay. That device should indeed be supported.

Is the driver loaded? (Does typing `lsmod | grep 8192cu` in a terminal produce any output?)

If so, what happens if you unload the driver (`sudo rmmod 8192cu`) and load it again (`sudo modprobe 8192cu`)? What do, say, the last 20 lines of the kernel log look like after you do that? (`dmesg | tail -20`)

concifederico, 2015-05-22:
Hi and thanks, here we go:

federico@H81M-H:~$ lsmod | grep 8192cu
8192cu                527317  0

federico@H81M-H:~$ sudo rmmod 8192cu
federico@H81M-H:~$ sudo modprobe 8192cu
federico@H81M-H:~$ dmesg | tail -20
[  419.222883] usb 3-1: new high-speed USB device number 114 using xhci_hcd
[  419.222950] usb 3-1: Device not responding to set address.
[  419.426812] usb 3-1: Device not responding to set address.
[  419.630709] usb 3-1: device not accepting address 114, error -71
[  420.606518] usb 3-1: new high-speed USB device number 119 using xhci_hcd
[  420.606584] usb 3-1: Device not responding to set address.
[  420.810452] usb 3-1: Device not responding to set address.
[  421.014295] usb 3-1: device not accepting address 119, error -71
[  421.878149] usb 3-1: new high-speed USB device number 123 using xhci_hcd
[  421.878215] usb 3-1: Device not responding to set address.
[  422.082098] usb 3-1: Device not responding to set address.
[  422.285978] usb 3-1: device not accepting address 123, error -71
[  423.045881] usb 3-1: new high-speed USB device number 127 using xhci_hcd
[  423.045946] usb 3-1: Device not responding to set address.
[  423.249799] usb 3-1: Device not responding to set address.
[  423.453698] usb 3-1: device not accepting address 127, error -71
[  423.997614] usb 3-1: new high-speed USB device number 5 using xhci_hcd
[  423.997681] usb 3-1: Device not responding to set address.
[  424.201507] usb 3-1: Device not responding to set address.
[  424.405450] usb 3-1: device not accepting address 5, error -71

That's all for now, seems there is an address issue.
Regards,

Federico

Scellow, 2015-11-19:
My 8188CUS works fine with this driver:

```
lsusb
Bus 001 Device 004: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
```

pvaret, 2017-10-14:
Hi,

Sorry for the late update. I did some digging and it doesn't look like the error message is caused by the driver. Instead it seems to be a suspend issue: people report having solved it by running `echo -1 >/sys/module/usbcore/parameters/autosuspend` as root. (That said, I'm not convinced the error is caused by your wifi device; it could be coming from any other USB device in your system.)

That said, now. The rtl8xxxu driver from the mainline kernel works pretty fine these days. That's what I'm using now myself. I recommend giving it a try. You may still have to troubleshoot NetworkManager, though: in particular, consider [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Disabling it made the rtlxxxu driver work well for me.

closed 2017-10-14 by pvaret