pvaret/rtl8192cu-fixes #42: Building on openwrt

83GTIGuy, 2015-05-25:
I'm looking for some advice on build this on openwrt.  I've created an openwrt Makefile so it gets built and loaded at boot, lsmod shows the module loaded but no device files are created.
/sys/class/net doesn't show a wlan0 device neither does ifconfig.
/sys/module/8192cu/  shows the device entries

Does this require any other rtlwifi components to work properly?

Panderine, 2015-09-09:
Yes it does, here are all the openwrt packages that you need in order to make this card "work". I have major issues with this and I believe are related to the driver package. If you manage to make it work, please let me know. I'd never build anything for openwrt so i don't know how to do it, but I believe is not that hard. 

wpad                     
kmod-scsi-core
kmod-usb-storage
kmod-crypto-hash
kmod-lib-crc16
block-mount

kmod-rtlwifi           
kmod-rtlwifi-usb      
kmod-rtl8192cu           
kmod-rtl8192c-common

crudeanu, 2016-01-05:
Hey 83GTIguy

Would you mind sharing the Makefile&co for the openwrt build?
I have a Lamobo R1 (with 8192cu) and am looking for a solution too.

Thanks in advance.

83GTIGuy, 2016-01-07:
Code didn't work. I ended up using the rt5390 usb wifi.  That one actually
works and the modules in openwrt are there already.   The problem I had was
integrating the kernel module in with the openwrt device creation
components.  I don't know if the 8192 is a better chipset, but I'm pleased
with the 5390.

pvaret, 2016-01-07:
Honestly, at this point, I'd avoid the 8192 like the plague based on the driver issues alone. My Belkin N300 works fine, but judging from the pile of issues that keep being filed against this repo, I appear to be one of the lucky few.

83GTIGuy, 2016-01-07:
I agree - the future looks bright since the driver has recently been
reworked and the new updated driver is now in mainline kernel.  The
drawback right now with that new driver is that it doesn't yet support
softap or monitor mode.

pvaret, 2016-01-09:
I didn't know that the in-kernel driver has been reworked. This is great news! Do you know in what version of the kernel this new driver is available?

83GTIGuy, 2016-01-10:
BTW  the new rtl8xxxu drivers have been merged into the next 4.4 kernel.

pvaret, 2016-01-10:
Splendid news! FREE AT LAST, FREE! I'm looking forward to being able to close this repo. :)

dlech, 2016-01-10:
https://github.com/torvalds/linux/blob/master/drivers/net/wireless/realtek/rtl8xxxu/Kconfig

pvaret, 2017-10-14:
As a late update: I switched to the rtl8xxxu driver myself, and it works fine! I still had to troubleshoot NetworkManager, though: in particular, [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460) made the rtlxxxu driver work well for me.

closed 2017-10-14 by pvaret