pvaret/rtl8192cu-fixes #44: TxPower and Regulatory Domain.

aaroneusthegreat, 2015-05-28:
Thanks in advance for any help you may provide. 

My Problem:
I have yet been unable to change the txpower for my Alfa AWUS036NHR which uses the Realtek RTL8188RU chipset. I tried the usual commands using ifconfig to put the interface down, set the regulatory domain with iw reg set, and then change the txpower with iwconfig and bring the interface back up. When I do this, either raising or lowering the txpower, no change is reflected in the output from iwconfig. Either iwconfig is reporting the txpower incorrectly, or it's stuck at the default of 20dbm. 

I realize that this driver base code from Realtek is lagging behind the current kernel versions, and I really appreciate the work you've done to port this to current kernels. Without your work, my card wouldn't be working as well as it is right now. Just wanted to let you know I appreciate the time you put in. 

What I would like to know, is first off if you know how to fix this with this driver, or which sections of the code contain the regulatory domain handling or txpower or both. I am pretty experienced in C programming, so I should be able to make the proper changes to make this do what I want it to. 

Also another possibility, is the regulatory domain hard coded into the firmware? and if so is it possible to modify the firmware?

pvaret, 2016-11-19:
It's not going to help much at all, but Jes Sorensen (author of the shiny new mainline rtl8xxxu driver) recently gave a talk about the Realtek drivers, and the slides are available online: https://www.linuxplumbersconf.org/2016/ocw//system/presentations/4089/original/2016-11-02-rtl8xxxu-presentation.pdf

pvaret, 2017-10-14:
As a late update: the rtl8xxxu driver from the mainline kernel works pretty fine these days. That's what I'm using now myself. You may still have to troubleshoot NetworkManager, though: in particular, consider [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Disabling it made the rtlxxxu driver work well for me.

closed 2017-10-14 by pvaret