pvaret/rtl8192cu-fixes #45: Wifi drop lubuntu 15.04 [3.19.0-18-generic]

PenguinIsrael, 2015-06-01:
Everytime while time pass after I've installed the driver from here plus the power fix, wifi keep drop and force me disconnect it then reconnect, only to find after 2 minutes it drops again.

Any fix soon?

concifederico, 2015-06-03:
Device name and chip inside?

matthewvalimaki, 2015-06-20:
I have similar problem. To me it seems like it is more random than exactly 2 minutes. I have Lenovo A720 all-in-one desktop with rtl8192cu: Chip version 0x10, rtl8192cu: Board Type 3.

Let me know if more details are needed.

PenguinIsrael, 2015-06-22:
the model is TL-WN822N. Im pretty the chip is rtl8192cu.

shahar@Sonic:~$ lsusb
Bus 001 Device 047: ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter

Sorry to response so late.

parthipanramesh, 2015-06-29:
I'm having the same problem with the AWUS030HNR v2. I has the rtl8192 chip.
I installed this driver on my Ubuntu 15.04 with the 8192cu-disable-power-management.conf.
The connection drops after a few mintues.

Log:
```
Jun 29 08:34:09 AzureWave NetworkManager[891]: <warn> Couldn't disconnect supplicant interface: This interface is not connected.
Jun 29 08:34:10 AzureWave NetworkManager[891]: <info> (wlan2): supplicant interface state: scanning -> inactive
Jun 29 08:34:13 AzureWave NetworkManager[891]: (NetworkManager:891): libnm-util-CRITICAL **: get_secret_flags: assertion 'is_secret_prop (setting, secret_name, error)' failed
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Activation (wlan2) Stage 1 of 5 (Device Prepare) scheduled...
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Activation (wlan2) Stage 1 of 5 (Device Prepare) started...
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> (wlan2): device state change: need-auth -> prepare (reason 'none') [60 40 0]
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Activation (wlan2) Stage 2 of 5 (Device Configure) scheduled...
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Activation (wlan2) Stage 1 of 5 (Device Prepare) complete.
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Activation (wlan2) Stage 2 of 5 (Device Configure) starting...
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> (wlan2): device state change: prepare -> config (reason 'none') [40 50 0]
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Activation (wlan2/wireless): connection 'Wifi' has security, and secrets exist.  No new secrets needed.
Jun 29 08:34:13 AzureWave NetworkManager[891]: <error> [1435559653.711231] [platform/wifi/wifi-utils-wext.c:184] wifi_wext_set_powersave(): (wlan2): error setting powersave 1
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Config: added 'ssid' value 'Wifi'
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Config: added 'scan_ssid' value '1'
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Config: added 'key_mgmt' value 'WPA-PSK'
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Config: added 'auth_alg' value 'OPEN'
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Config: added 'psk' value '<omitted>'
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Activation (wlan2) Stage 2 of 5 (Device Configure) complete.
Jun 29 08:34:13 AzureWave NetworkManager[891]: <info> Config: set interface ap_scan to 1
Jun 29 08:34:14 AzureWave NetworkManager[891]: <info> (wlan2): supplicant interface state: inactive -> scanning
Jun 29 08:34:15 AzureWave wpa_supplicant[2382]: wlan2: Trying to associate with ac:11:14:12:ab:44 (SSID='Wifi' freq=2457 MHz)
Jun 29 08:34:15 AzureWave wpa_supplicant[2382]: wlan2: Association request to the driver failed
Jun 29 08:34:15 AzureWave NetworkManager[891]: <info> (wlan2): supplicant interface state: scanning -> associating
Jun 29 08:34:17 AzureWave wpa_supplicant[2382]: wlan2: CTRL-EVENT-DISCONNECTED bssid=ac:11:14:12:ab:44 reason=0
Jun 29 08:34:17 AzureWave wpa_supplicant[2382]: wlan2: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="Wifi" auth_failures=1 duration=10
Jun 29 08:34:17 AzureWave NetworkManager[891]: <info> (wlan2): supplicant interface state: associating -> disconnected
Jun 29 08:34:17 AzureWave NetworkManager[891]: <info> (wlan2): supplicant interface state: disconnected -> scanning
```

Any idea why this is happening?

pvaret, 2015-07-09:
Noted. Afraid I don't know why this is happening, no. Note that some people have reported the exact same problem with the in-kernel driver, too, oddly.

PenguinIsrael, 2015-07-12:
I think its related to the nvidia driver too. I installed 304-76

mdh34, 2015-11-26:
I had this problem as well, it could simply be that you don't have enough signal or that the antennae are not strong enough - I balanced a metal screwdriver on top of mine to fix the issue!

pvaret, 2017-10-14:
I experienced and managed to fix that very issue recently. It was a power management problem and the fix was to [disable power management in NetworkManager](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Hopefully this will fix it for other people too. As a side note, consider switching to the rtl8xxxu driver from the mainline kernel, which works pretty fine these days.

closed 2017-10-14 by pvaret