pvaret/rtl8192cu-fixes #46: Does anyone have this working on 4.0.5?

Pastafarianist, 2015-07-13:
I'm not really sure when the driver started misbehaving, but I think it happened approximately at the time when I removed the old kernel sources from my system and started upgrading to the new kernel. In any case, right now I'm running 4.0.5 and the driver isn't working well. Specifically, [here's a log of pinging my router](https://bpaste.net/show/961abf6af34e). As you may see, there are intermittent periods where the round trip time is measured in seconds. The driver's kernel module [is loaded](https://bpaste.net/show/b3dcaf683f1a).

So, does anyone have this working on 4.0.5? And if yes and it's working, how can I troubleshoot the issue?

pvaret, 2015-07-27:
Not tested it on kernels that recent. I doubt I will have time to see to that in the near future.

Getting the actual, in-kernel driver fixed is really becoming urgent...

paullaz, 2015-07-31:
I am running kernel 4.1.3 on a Fedora and the ping is consistent at 5-6ms so far.

Pastafarianist, 2015-08-11:
The behavior is erratic and I can't figure out when it works and when it doesn't. Here's another piece of stats, a graph of file copy speed over Samba:

[image]

rZeton, 2016-01-03:
I built fresh 4.3.3 kernel from https://www.kernel.org/ and it works ok with it. I have power management setting copied over as well.

http://www.dslreports.com/tools/pingtest

pvaret, 2016-01-04:
I since upgraded to kernel 4.2.0 and the driver works as expected here too. Whatever your issue is, @Pastafarianist, it is probably not related to the kernel version. The likeliest explanation is that your device is one of those that the driver just doesn't support very well. And if so, sadly, I don't think there is much we can do about it.

Pastafarianist, 2016-01-04:
In fact, a couple of months ago I got tired of laggy internet connection and, being unable to resolve this issue, I moved to Ethernet.

My device is TP-LINK TL-WN823N.

pvaret, 2016-01-04:
Well, fiddlesticks. This device is on the list of devices reported to work. So either those reports were overly optimistic, or something changed in recent kernels and the devices in question stopped working (but others somehow still run fine). This sucks.

JackShannon, 2016-01-24:
I have the same device and have not been able to get it to work, has anyone been able to?

Pastafarianist, 2016-01-26:
I have tested the device with a fresh installation of Lubuntu 14.04.3 and saw similar drops in round trip time and lost packets. The problem is most likely not specific to this driver: it's either in the kernel or in the device.

pvaret, 2017-10-14:
I experienced and managed to fix that very issue recently. It was a power management problem and the fix was to [disable power management in NetworkManager](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Hopefully this will fix it for other people too. As a side note, consider switching to the rtl8xxxu driver from the mainline kernel, which works pretty fine these days.

closed 2017-10-14 by pvaret