pvaret/rtl8192cu-fixes #47: does not work with AWUS036NHR

oliverkurth, 2015-07-18:
I have an Alfa AWUS036NHR usb wifi adapter, and tried this driver in a Ubuntu 15.04 VM with kernel 4.0.0. I also tried the driver on the RPi, with the same issues. An Edimax dongle works fine in both the RPi and Ubuntu VM with the same driver, both in managed mode and as an AP on the RPi (haven't tested AP mode in Ubuntu yet).

Setting the essid does not work:

vmware@ubuntu:~/rtl8192cu-fixes$ sudo ifconfig wlan1 up
vmware@ubuntu:~/rtl8192cu-fixes$ sudo iwconfig wlan1 essid fubar
vmware@ubuntu:~/rtl8192cu-fixes$ iwconfig wlan1
wlan1     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.437 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

And wpa_supplicant does not work either:

vmware@ubuntu:~$ sudo wpa_supplicant -i wlan1 -c /etc/wpa_supplicant.conf -Dnl80211,wext
Successfully initialized wpa_supplicant
nl80211: Driver does not support authentication/association or connect commands
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
wlan1: Trying to associate with c0:3f:0e:9a:xx:xx (SSID='philatri' freq=2462 MHz)
wlan1: Association request to the driver failed
wlan1: CTRL-EVENT-DISCONNECTED bssid=c0:3f:0e:9a:xx:xx reason=0
wlan1: Trying to associate with c0:3f:0e:9a:xx:xx (SSID='philatri' freq=2462 MHz)
wlan1: Association request to the driver failed

Kernel is 4.0.0:

Linux ubuntu 4.0.0-040000-generic #201504121935 SMP Sun Apr 12 23:36:33 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

My goal is to use the Alfa as an AP on the RPi, but I use the Ubuntu VM for testing.

pvaret, 2015-07-27:
Hi,

Do you have the USB id of the device? (You can obtain it with the command `lsusb`.)

xXx-stalin-666-money-xXx, 2015-08-30:
Hello, i have the same adapter, same problem - USB id is 0bda:817f 
Also, some more info: airmon-ng (aircrack) can't see the device, but iwconfig can.
It is possible to connect to unsecured networks, but WEP/WPA doesn't work. I also tried this driver -same results: https://github.com/dz0ny/rt8192cu

pvaret, 2015-09-03:
Ok, thanks for the update. It really seems to be hit-and-miss as to whether any given device will work. I yet have to work out why.

tobiasdiez, 2015-11-10:
I have the same adapter (same USB id as @xXx-stalin-666-money-xXx) and the same problem. Scanning works (for example with wifi-menu) but as soon as I want to connect to a network with wpa security, I run into problems.
Any update on this issue?

pvaret, 2015-11-11:
Sadly, no. It bears repeating -- this repository is a repackaging of Realtek's own driver, with the purpose of making it available on recent Ubuntu versions; but if Realtek won't fix a given issue, odds are very low it will be fixed in this repository. (Sorry!) And I'm afraid I do not know why some devices just don't work with it.

Nik1990, 2015-11-27:
Any news? I've the some problem, AWUS036NHR (no v.2) with Ubuntu 14.04, kernel 3.19.

boourns, 2015-11-28:
exact same problem as listed (subscribing to the issue in case someone figures it out :) )

pvaret, 2017-10-14:
As a late update: the rtl8xxxu driver from the mainline kernel works pretty fine these days. That's what I'm using now myself. You may still have to troubleshoot NetworkManager, though: in particular, consider [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Disabling it made the rtlxxxu driver work well for me.

closed 2017-10-14 by pvaret