pvaret/rtl8192cu-fixes #49: compile error

Tomius, 2015-08-07:
For me the compilation fails, and I get the following log file:
http://pastebin.com/wS3EY6Sg

Tomius, 2015-08-07:
(I guess my compiler doesn't treat functions marked with __inline as inline functions)

pvaret, 2015-09-03:
Hi Tomius, which compiler is that?

Tomius, 2015-09-03:
gcc 5.1.0

$ cc --version
cc (Ubuntu 5.1.0-0ubuntu11~14.04.1) 5.1.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pvaret, 2015-11-11:
Hi Tomius,

Welp, I know it won't help, but I can confirm that the driver builds fine on every version of Ubuntu since 14.04. Could there be something specific to your setup that causes the breakage?

Tomius, 2015-11-11:
I can't think of any setup, that could cause my compiler to ignore __inline specifications, without a single warning. I'm sure that the LTS ubuntu (*.04) has a different compiler version than mine (*.10) However, if it only applies to me, then this error is not relevant - it took about 2 mins to rewrite the __inline functions to be macros, and it worked that way.

closed 2015-11-11 by Tomius

pvaret, 2015-11-11:
I have not had other reports about this specific issue, so I'd be cautiously tempted to say it's indeed only you. Although I'm afraid I have no idea where that could come from!