pvaret/rtl8192cu-fixes #50: Cannot install 8192cu/1.10

Caustiticus, 2015-08-09:
Great work you're doing here. Sadly, I'm stuck at the moment.

Trying to compile this on a RasPi2 with Raspbian (should have just bought that silly Edimax and been done with it, but noooo~). Fairly new to any editing of Linux beyond adding stuff and compiling things.

Followed every step by the numbers after some back-tracking to get the necessary packages installed ( included updating git ), and everything was going smoothly until the actual install step; then I get this error:

pi@raspberrypi / $ sudo dkms install 8192cu/1.10
Error! echo
Your kernel headers for kernel 3.18.11-v7+ cannot be found at
/lib/modules/3.18.11-v7+/build or /lib/modules/3.18.11-v7+/source.

Tried several times to figure out what was going wrong, but can't. cd'd into /lib/modules/3.18.11-v7+ and there are no "build" or "source" directories, as shown:

pi@raspberrypi /lib/modules/3.18.11-v7+ $ ls
kernel             modules.builtin      modules.dep.bin  modules.softdep
modules.alias      modules.builtin.bin  modules.devname  modules.symbols
modules.alias.bin  modules.dep          modules.order    modules.symbols.bin

 Have no idea what's up. Can I get some help here?

dlech, 2015-08-09:
The raspberry pi kernel already includes this driver, so you don't need to compile it yourself.

https://github.com/raspberrypi/linux/tree/rpi-3.18.y/drivers/net/wireless/rtl8192cu

leewc, 2015-08-09:
@dlech  I think the raspberry pi kernel's driver isn't exactly functional. It's also mentioned in the wiki for Arch Linux that one should try the dkms version if the included drivers don't work, https://wiki.archlinux.org/index.php/Wireless_network_configuration#rtl8192cu

That however, has compile errors, a string method being deprecated in `os_dep/linux/rtw_android.c`, which this fork fixes nicely. I got it to install after cloning by doing a `make dkms` after `cd rt8192cu`, and then a `depmod -a` to refresh the drivers. Still couldn't connect to wifi though.

dlech, 2015-08-09:
I believe the comment on the arch linux wiki was referring to the mainline kernel driver. Since the mainline driver doesn't work very well, the raspberry pi folks decided to include the same driver that is used in this repository instead of the mainline kernel driver in the raspberry pi kernel.

leewc, 2015-08-10:
@dlech  I see. I can confirm the driver works much better on Raspbian (at time of writing it is the latest `2015-05-05-raspbian-wheezy` image). Pretty much the main reason why I went from ARM Arch Linux to Raspbian. Pretty much plug and play without the need to compile this driver.

Caustiticus, 2015-08-10:
@dlech Well I know for a fact that on MINE (running NOOBS Raspbian, recently updated post-install) that the driver isn't working. I plug it in and a red light pops on the WiFi adapter for a second, then nothing. Checked ```ifconfig -a```, and it IS reading it, but didn't give it any addresses or anything. I've tried half of every trick I can find (some of which flat-out don't work b/c of differences between distros), and nothing has worked so far. 

Even booted up into the GUI, and when I look at the WiFi settings, it only gives me the option to switch to SSID, and everything else is grayed out. So yeah, I'm trying this method now. And from what I've heard the built-in WiFi drivers for Raspbian might be kinda crummy anyhow :/

@leewc At this rate I may just reflash and use that distro ~.~ 
Raspbian is easy to use and definitely built for beginners, but is quickly becoming a PITA due to being lightweight package-wise, and importing packages and patching workarounds is becoming a chore. The only thing I have installed right now is Moonlight (an open-source version of nVidia's GeForce Experience GameStream program), and I've already reinstalled that a few times due to "complications". :P Thanks for the tip.

The USB WiFi I'm using is a Kootek "Wireless-N USB Adapter" which apparently uses the Realtek RTL8192CU driver. Got it instead of the highly-recommended Edimax because it said it had 300Mbit/s instead of 150Mbit/s for the same price, but I'm wondering if I shouldn't have gone for the Edimax now :/

dlech, 2015-08-10:
@Caustiticus, we would need to know the output of `lsusb` to know exactly what chip is in there, but according to [this](http://www.amazon.com/Kootek-Raspberry-Wifi-Dongle-Adapter/dp/B00FWMEFES), the Kootek is RTL8188CUS, not rtl8192cu, so that may explain why it is not working.

Caustiticus, 2015-08-10:
@dlech Well then the package and store front (bought off Amazon.com) are bold-facedly lying to me :\ 
On the adapter's package: "(find the drivers for [b]RTL8192CU[/b] to download)"
On the back: "Chipset: Realtek RTL8192CUS"

And I swear ```lsusb``` outputed the same last time I tried it. Will try again and edit the results in.

dlech, 2015-08-10:
The USB ID `0bda:8178` is in the 8192cu driver, so it should work. (Searching indicates that RTL8912CU and RTL8188CUS are basically interchangeable)

What is output of `lsmod`? `8192cu` or `rtl8192cu`?

Caustiticus, 2015-08-10:
pi@raspberrypi ~ $ lsmod
    Module                  Size  Used by
    cfg80211              420690  0
    rfkill                 16659  1 cfg80211
    snd_bcm2835            19769  0
    snd_pcm                74825  1 snd_bcm2835
    snd_seq                53561  0
    snd_seq_device          3650  1 snd_seq
    snd_timer              18157  2 snd_pcm,snd_seq
    snd                    52116  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
    **8192cu**                528485  0
    joydev                  9047  0
    evdev                  10232  4
    uio_pdrv_genirq         2966  0
    uio                     8235  1 uio_pdrv_genirq

I assume that one is the basic driver?
Only one I see is 8192cu. :/

dlech, 2015-08-10:
`8192cu` is what you would get if you compiled the code in this repository (which as I said is already in the RPi kernel). `rtl8192cu` is the mainline kernel driver, which is what I am guessing that you mean by "basic".

pvaret, 2015-09-03:
Causticus, would you consider your problem solved? If the driver that ships with the RPi kernel indeed works, I should probably make a note of that in the README.

dlech, thanks a thousand for help, by the way. :)

Caustiticus, 2015-09-08:
@pvaret Yeah, I'll have to start over to see if something got screwed sideways unintentionally (still new to this darn silly thing XD), but I think a different approach is necessary for a RasPi. Closing thread.

closed 2015-09-08 by Caustiticus