pvaret/rtl8192cu-fixes #52: max speed is 54mbit?

FPeter84, 2015-09-22:
Hi!

First of all, I would like to say big thank You for efforts! It looks Your modification solved the long-existing problem of onboard wifi module on BananaPi / Lamobo R1 board! It was absolutely unusable / unstable with current images (both openwrt, raspbian, etc) but now it looks stable and reliable!

The only problem is the low speed: Im unable to get more than theoretical 54mbit and 24-28mbit measured in practice (iperf3)... I tried it with both WPA2 and unsecured mode, I checked it with several clients (similar RTL8188CU, AR9285, etc), both windows and ubuntu and the result is the same...

CPU and memory load is low during transmission (Cortex-A7 dualcore 1GHz, 1GB RAM). My current environment is Ubuntu 14.04.3 LTS with kernel 4.2.0 from armbian.com/lamobo-r1

Do You have any idea, what is the problem? Or is it limited in the driver? Based on informations from Realtek's homepage, it should be able to handle theoretically 300Mbps in 802.11n mode...

Best regards,
Peter

dcds, 2015-10-04:
Hi, I also like to thank you very much for the driver! Nevertheless I have the same issue, the connection speed is very slow. So I would also like to know if somebody had the same issue, if this a bug, or something that can be solved through configuration tuning.

Best Regards,
Daniel

pvaret, 2015-11-11:
Hi,

Thank you for the kind words!

Honestly, I didn't even know this driver worked on the RPi and friends, so that's good to hear. It bears repeating, though, this repository only contains a repackaging of Realtek's own driver to make it easier to install on Ubuntu (though I'm glad it works in other distros). If something is broken in Realtek's driver, it will be broken here. Sorry. :/

closed 2015-11-11 by pvaret