pvaret/rtl8192cu-fixes #55: Configure as access point mode / master mode NOT successful

sean78901, 2015-10-09:
Hi, thank you for your work.
I followed the instructions and the installation of the 8192cu fix is successful because I can connect wirelessly to the internet. However, I am unable to configure the 8192cu as an Access Point for my other devices to access the internet. In particular, I am unsuccessful in setting the 8192cu mode to either ap or to master (see questions (3) and (4) below respectively).

Ubuntu 14.04
ID 0b05:17ba ASUSTek Computer, Inc. (Asus USB-N10 Nano Wireless-N USB Adapter)

(1) Does your module support my hardware / Is my device ID on the supported list?

(2) lsmod | grep 8192cu gives:
8192cu                527333  0
Is this a problem too?

(3) To check whether AP mode is supported, I tried iw list and iw dev but the interface / device does not show up, and therefore I am unable to confirm that the 8192cu supports AP mode.
Is the 8192cu fix under the mac80211 framework?
If so, then wouldn't iw list not showing the 8192cu be a problem?

(4) To change mode to master, I tried sudo iwconfig wlan1 mode master .
sudo iwconfig confirms that the mode has been changed to master , but the moment I use Network Manager to connect, the mode reverts back to managed .

(5) I tried the instructions in the following URL (particularly the last step of Step 2) to configure the 8192cu as an Access Point, but to no avail - 
http://askubuntu.com/questions/180733/how-to-setup-an-access-point-mode-wi-fi-hotspot

(6) Do I have to use hostapd or would the instructions in the URL above (which is a simple hack) work for the 8192cu to be configured as an Access Point?

Appreciate your assistance. Thank you.

dlech, 2015-10-09:
Duplicate of #27.

sean78901, 2015-10-09:
Hi dlech,
Thank you for the comment.
I saw #27 before I posted for help, but dismissed it as I didn't understand what the link back to realtek meant. On second read, was your comment in #27 referring to a solution akin to this - 

http://wannabe-nerd.tweakblogs.net/blog/10870/wifi-access-point-using-a-realtek-8192cu-based-usb-wifi-dongle-with-a-raspberry-pi.html
?

If yes, then, of the following parts of the solution (in the link above) - 
(1) Installation and configuration of bridging
(2) Install kernel sources
(3) Download sources and compiling the rt8192cu driver / kernel module
(4) Download sources for hostapd and compile them
(5) Configuration ,
which parts can I skip?

Can I just do the steps in parts (1), (4) and (5) and skip (3) given that I have already installed the 8192cu fix provided by pvaret?
What about part (2) - can I skip part (2)?

Thank you.

dlech, 2015-10-09:
Yes, you can skip part (2) - it is only needed for raspberry pi. And yes, if you have already installed the 8192cu driver, then you don't have to do step (3) again.

sean78901, 2015-10-10:
Hi dlech,
Thank you once again.
I am having a problem that I believe is not related to the 8192cu driver fix, but for completeness, I shall describe my problem here even as I seek an answer at an appropriate forum. Basically I am trying to accomplish the following setup - 
wireless internet ---> wlan0 of Ubuntu computer "bridged to" wlan1 of same Ubuntu computer ---> device with wifi

The reason for this setup is that the wireless internet is a WPA2 Enterprise network which my device with wifi cannot connect to. (The device with wifi also cannot connect to peer-to-peer networks or ad-hoc networks.)
wlan0 is an Intel wireless chipset / interface which cannot be set to either access point mode or master mode, and therefore is used to connect to the WPA2 Enterprise internet source.
wlan1 is the Asus / RealTek 8192cu installed with pvaret's fix and is to be set to access point mode.
Obviously, the "bridge" between wlan0 and wan1 is not meant to be wireless.

(1) I believe brctl cannot bridge a wireless interface (wlan0) to another wireless interface (wlan1).
Is this correct?
(2) If brctl cannot be used, can I still use hostapd?
(3) If brctl cannot be used, what should I be looking for to accomplish the setup described above?
(4) Would any of the following be a part of the solution to my problem - 
(a) dnsmasq to serve DHCP
(b) NAT (Network Address Translation)
(c) iptables
(d) Linux IP Masquerade feature
(e) sudo iw set dev wlan0 4addr on
(f) WDS (wireless distribution system) ?

Thank you.

sean78901, 2015-10-16:
[Solved]
I used iptables instead of brctl. Issue closed.

closed 2015-10-16 by sean78901