pvaret/rtl8192cu-fixes #56: Doesn't work on Ubuntu 15.10

srokatonie, 2015-10-29:
After you follow all the steps wifi network list is gone.

SamuelVil, 2015-11-11:
Same here.  Since I've followed the steps, then it's just like my TL-WN822N adapter wasn't plugged in...

jeka-kiselyov, 2015-11-16:
Same for me.

dlech, 2015-11-16:
So, everything compiles OK, but no adapter appears?

Can someone post the output of `dmesg` from when you plug in your adapter? 

Ubuntu 15.10 uses 3.19.x kernel correct?

jeka-kiselyov, 2015-11-16:
@dlech 

No, it's:
```
Linux jeka911-desktop 4.2.0-18-generic #22-Ubuntu SMP Fri Nov 6 18:25:50 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

I believe this is an issue?

dmesg results:
```
[   81.984947] usb 6-2: new high-speed USB device number 2 using ehci-pci
[   82.123386] usb 6-2: New USB device found, idVendor=7392, idProduct=7811
[   82.123395] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   82.123400] usb 6-2: Product: 802.11n WLAN Adapter
[   82.123405] usb 6-2: Manufacturer: Realtek
[   82.123409] usb 6-2: SerialNumber: 00e04c000001
[   83.154710] 8192cu: module verification failed: signature and/or required key missing - tainting kernel
[   83.309182] usbcore: registered new interface driver rtl8192cu
[   83.321548] rtl8192cu 6-2:1.0 wlx74da3854880a: renamed from wlan0
[   83.359145] cfg80211: World regulatory domain updated:
[   83.359154] cfg80211:  DFS Master region: unset
[   83.359158] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   83.359163] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   83.359168] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   83.359172] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   83.359176] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   83.359181] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   83.359185] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   83.359189] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   83.359193] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
```

Network manager is empty after this. There's even no "Enable Wi-Fi" menu item. Switched back to original rtl8192cu, by removing blacklist file from /etc/modprobe.d/ and rebooting.