pvaret/rtl8192cu-fixes #57: No connection after wake up

ayrton04, 2015-11-01:
I'm using a TRENDnet TEW-648UBM N150 Micro Wireless Adapter, and your driver works very well (thank you!), except when I put my computer to sleep and wake it back up. When it wakes up, the adapter either doesn't show up in the network manager drop-down in the menu bar, or there are no networks listed. Rebooting fixes the problem, and the adapter connects immediately. Relevant system specs:

- uname -a output: Linux Ayrbox 3.19.0-25-generic #26~14.04.1-Ubuntu SMP Fri Jul 24 21:16:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux]
- ASRock X99 Extreme4 LGA 2011-v3 Intel X99 motherboard
- TRENDnet TEW-648UBM N150 USB WiFi adapter

Apologies for posting a new issue if there's just a configuration change that I need to make.

pvaret, 2015-11-11:
Hi,

Glad to hear the driver works well for you! Have you tried the power management workaround listed in the README? (I don't think it will change anything, but you never know.)

ayrton04, 2015-11-11:
Whoops, that did in fact do the trick, sorry. Thanks very much!

closed 2015-11-11 by ayrton04

pvaret, 2015-11-11:
Oh, but you're welcome! Sorry about the delay in replying, though. I have not had much time to take care of this repository lately.