pvaret/rtl8192cu-fixes #59: ASUS USB-N 13 work

shannara, 2015-11-15:
Hi,

I write this that usb wifi key work perfectly with rtl8192cu-fixes.
It's require for rate up to 300Mb/s, limit to 150Mb/s with default module.

Thanks a lot.

dlech, 2015-11-15:
Thanks for sharing. Can you post the output of `lsusb` for this device and we will add it to the readme.

shannara, 2015-11-21:
Hi,
output of lsusb

Bus 001 Device 002: ID 0b05:17ab ASUSTek Computer, Inc. USB-N13 802.11n Network Adapter (rev. B1) [Realtek RTL8192CU]

Sorry for delay.

HashBreakerPC, 2017-05-10:
on Linux 4.9.0-kali4-amd64 #1 SMP Debian 4.9.25-1kali1 (2017-05-04) x86_64 GNU/Linux
`Bus 002 Device 003: ID 0b05:17ab ASUSTek Computer, Inc. USB-N13 802.11n Network Adapter (rev. B1) [Realtek RTL8192CU]`
iw dev not show my device
```
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr fe:93:dc:d4:e1:61
		type managed
		txpower 16.00 dBm
```
Navigation is perfect but aircrack does not detect the device
```
wlan1     IEEE 802.11bgn  ESSID:"MEDIACOM"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.462 GHz  Access Point: C8:3A:35:06:3A:D0   
          Bit Rate:300 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=97/100  Signal level=62/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
```

closed 2017-10-14 by pvaret