pvaret/rtl8192cu-fixes #60: 8188CUS unable to enable monitor mode

Scellow, 2015-11-19:
Hello

I'm trying to enable monitor mode using the following command but it just doesnt work

sudo airmon-ng start wlx000b81857f75

Found 5 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to kill (some of) them!

PID	Name
6636	NetworkManager
6644	wpa_supplicant
6648	dhclient
6844	avahi-daemon
6847	avahi-daemon
Process with PID 6648 (dhclient) is running on interface wlx000b81857f75

Interface	Chipset		Driver

wlx000b81857f75		Unknown 		rtl8192cu (monitor mode enabled)

Ok so i try to kill these processes, but then unable to find that interface, i try to manually enable it with:

ifconfig wlx000b81857f75 up 

But nothing

I'm missing something?

Scellow, 2015-11-19:
Forgot to add device info:

```
Bus 001 Device 004: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
```

dlech, 2015-11-20:
Duplicate of #10 (and #22)?

Scellow, 2015-11-21:
Oh yes you are right!

I might try the other driver to see if it's fixed

Thanks a lot for the effort, this driver is awesome, despite the little monitor mode problem

closed 2015-11-21 by Scellow