pvaret/rtl8192cu-fixes #63: Error! Bad return status for module build on kernel: 3.18.16-v7 (armv7l)

cbogithub, 2016-01-09:
root@kali:~# sudo dkms install 8192cu/1.10

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=3.18.16-v7 -C /lib/modules/3.18.16-v7/build M=/var/lib/dkms/8192cu/1.10/build......(bad exit status: 2)
Error! Bad return status for module build on kernel: 3.18.16-v7 (armv7l)
Consult /var/lib/dkms/8192cu/1.10/build/make.log for more information.

pvaret, 2016-01-09:
Well then, what does /var/lib/dkms/8192cu/1.10/build/make.log say?

cbogithub, 2016-01-09:
root@kali:~# cat /var/lib/dkms/8192cu/1.10/build
cat: /var/lib/dkms/8192cu/1.10/build: Is a directory
root@kali:~# cat /var/lib/dkms/8192cu/1.10/build/make.log
DKMS make.log for 8192cu-1.10 for kernel 3.18.16-v7 (armv7l)
Sat Jan  9 11:02:19 UTC 2016
make: Entering directory '/usr/src/kernel'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  LD      /var/lib/dkms/8192cu/1.10/build/built-in.o
  CC [M]  /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o
./scripts/recordmcount: 1: ./scripts/recordmcount: Syntax error: "(" unexpected
scripts/Makefile.build:257: recipe for target '/var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o' failed
make[1]: *** [/var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o] Error 2
Makefile:1381: recipe for target '_module_/var/lib/dkms/8192cu/1.10/build' failed
make: *** [_module_/var/lib/dkms/8192cu/1.10/build] Error 2
make: Leaving directory '/usr/src/kernel'

dlech, 2016-01-09:
> 3.18.16-v7

This sounds like a Raspberry Pi kernel. The [Raspberry Pi kernel](https://github.com/raspberrypi/linux) already has this driver patched into it. No need to use the source from this repository. See #50.

closed 2017-10-14 by pvaret