pvaret/rtl8192cu-fixes #67: crash on  debian 3.2.0-4-powerpc

marianpopeanga, 2016-02-12:
This is what i get when i try to use this driver on Debian power pc 32 bit with kernel 
3.2.0.4 . I just compiled the version from here. When i plug in the wifi goungle i get this error. 
Also i get the same error is i use the driver frm edimax.com web site, even through there are differences in size of these drivers, they report same version:

filename:       /lib/modules/3.2.0-4-powerpc/kernel/drivers/net/wireless/8192cu.ko
version:        v4.0.2_9000.20130911
description:    Realtek Wireless Lan Driver
srcversion:     C657B8B3541F2E8A45866D2
vermagic:       3.2.0-4-powerpc mod_unload modversions 
parm:           rtw_chip_version:int

  445.615348] udevd[362]: renamed network interface wlan0 to wlan4
[  446.205825] ADDRCONF(NETDEV_UP): wlan4: link is not ready
[  447.565464] Unable to handle kernel paging request for instruction fetch
[  447.565480] Faulting instruction address: 0x00000000
[  447.565489] Oops: Kernel access of bad area, sig: 11 [#1]
[  447.565494] PowerMac
[  447.565499] Modules linked in: 8192cu(O) nls_utf8 udf cpufreq_conservative cpufreq_powersave cpufreq_ondemand cpufreq_userspace cpufreq_stats radeon parport_pc lp parport power_supply ttm drm_kms_helper drm bnep rfcomm bluetooth rfkill crc16 binfmt_misc fuse nfsd nfs nfs_acl auth_rpcgss fscache lockd sunrpc reiserfs snd_powermac loop firewire_sbp2 snd_aoa_codec_toonie snd_aoa_fabric_layout snd_aoa snd_aoa_i2sbus snd_aoa_soundbus snd_pcm snd_page_alloc snd_seq i2c_powermac snd_seq_device snd_timer joydev evdev snd soundcore ext3 mbcache jbd dm_mod usbhid hid sg sr_mod firewire_ohci ohci_hcd ehci_hcd firewire_core usbcore sungem cdrom sd_mod crc_itu_t sungem_phy crc_t10dif usb_common [last unloaded: scsi_wait_scan]
[  447.565597] NIP: 00000000 LR: f3cd7278 CTR: 00000000
[  447.565605] REGS: edde7eb0 TRAP: 0400   Tainted: G           O  (3.2.0-4-powerpc Debian 3.2.73-2+deb7u2)
[  447.565611] MSR: 40009032 <EE,ME,IR,DR>  CR: 84000082  XER: 00000000
[  447.565624] TASK = edcbb1f0[3704] 'RTW_CMD_THREAD' THREAD: edde6000
[  447.565628] GPR00: 00000000 edde7f60 edcbb1f0 f3d4d000 c1f3aa08 00000000 c1f3aa08 00040966 
[  447.565641] GPR08: 00000000 f3d10000 00000001 00000020 c001c9a8 00000000 01400000 02400000 
[  447.565653] GPR16: 01860d50 018608e0 0191fdc0 ffbbf400 018604e4 01860b82 019120e8 f3d4e4ac 
[  447.565665] GPR24: 00000004 f3d4e4a0 c1f3aa00 00000003 f3d155b8 f3d4d000 c1f3aa00 f3d127c0 
[  447.565678] NIP [00000000]   (null)
[  447.565754] LR [f3cd7278] mlme_evt_hdl+0x80/0xa4 [8192cu]
[  447.565758] Call Trace:
[  447.565781] [edde7f60] [f3cbf590] rtw_free_cmd_obj+0x5c/0x70 [8192cu] (unreliable)
[  447.565806] [edde7f80] [f3cbfb5c] rtw_cmd_thread+0x1bc/0x260 [8192cu]
[  447.565823] [edde7fb0] [c005d808] kthread+0x78/0x7c
[  447.565836] [edde7ff0] [c000f524] kernel_thread+0x4c/0x68
[  447.565840] Instruction dump:
[  447.565845] XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX 
[  447.565855] XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX XXXXXXXX 
[  447.565869] ---[ end trace 71157fa58523be13 ]---

dlech, 2016-02-12:
The fixes in this repository are not needed for kernels < 3.8 or so. Try using the driver provided by realtek. I would give a link, but their website seems to be down at the moment.

marianpopeanga, 2016-02-12:
I get the same error using the driver on they web site. It seems the code is shared with that one.
Probably i should also notify them about this error. Do you know where?

marianpopeanga, 2016-02-12:
I mean i get the same error with the driver from edimax.com. The web site from realtek is down for some time. Do anyone here has that driver from realtek already download  to give it a try ?

pvaret, 2016-02-15:
Yes, this smells like it should probably be reported to RealTek. Edimax
just makes products that use RealTek's chipsets, so probably can't do
anything about it.

marianpopeanga, 2016-02-16:
They have their own driver, but looks untested on this platform

closed 2017-10-14 by pvaret