pvaret/rtl8192cu-fixes #68: How to uninstall

davidernst9, 2016-02-18:
I have switched my wifi usb stick in the meantime since I installed this driver. I would want to uninstall this custom driver and to regain the default ubuntu 15.10 state as before the installation but I do not know how to.

Is this possible without reinstalling the whole OS? It would be nice to include such instructions (or a link to an applicable explanation) below the installation guidelines.

dlech, 2016-02-18:
If you followed the instructions in the readme, then just uninstall using [dmks](http://manpages.ubuntu.com/manpages/wily/en/man8/dkms.8.html).

    sudo dkms uninstall 8192cu

And if you copied the modprobe blacklist

    sudo rm /etc/modprobe.d/rtl8192cu-fixes/blacklist-native-rtl8192.conf

davidernst9, 2016-02-18:
Thank you for the quick response.
However, there seems to be another problem. If I understand this 
correctly I need to specify the version number before I can uninstall 
the driver.

david@david-desktop:~$ sudo dkms uninstall 8192cu
Error! Invalid number of arguments passed.
Usage: uninstall <module>/<module-version> or
        uninstall -m <module>/<module-version> or
        uninstall -m <module> -v <module-version>

Best regards,

David Ernst

dlech, 2016-02-18:
This is why I provided a link to the man page - in case I missed something. Did you try this?

    sudo dkms uninstall 8192cu/1.0

If that is the wrong version, read the man page. It tells you where files are stored and you can figure out what version you used.

pvaret, 2016-02-18:
Hi @davidernst9,

The version is the one used during the installation, that is, if the command you used in order to install the module was:

    sudo dkms install 8192cu/1.10

Then the uninstall command is:

    sudo dkms uninstall 8192cu/1.10

1.10 is the latest version. Find out which version(s) you have installed with:

    dkms status 8192cu | grep installed

closed 2016-02-18 by pvaret