pvaret/rtl8192cu-fixes #69: Arch Linux Rpi Fail to Make

bass0324, 2016-02-24:
Attempted to make this with the following in Makefile configuration for the Raspberry Pi Model B
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT

Error Output:
make[1]: Entering directory '/usr/lib/modules/4.1.18-1-ARCH/build'
  CC [M]  /home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.o
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c: In function 'rtw_cfg80211_indicate_disconnect':
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:703:11: error: 'struct wireless_dev' has no member named 'sme_state'
   if(pwdev->sme_state==CFG80211_SME_CONNECTING)
           ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:703:24: error: 'CFG80211_SME_CONNECTING' undeclared (first use in this function)
   if(pwdev->sme_state==CFG80211_SME_CONNECTING)
                        ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:703:24: note: each undeclared identifier is reported only once for each function it appears in
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:706:16: error: 'struct wireless_dev' has no member named 'sme_state'
   else if(pwdev->sme_state==CFG80211_SME_CONNECTED)
                ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:706:29: error: 'CFG80211_SME_CONNECTED' undeclared (first use in this function)
   else if(pwdev->sme_state==CFG80211_SME_CONNECTED)
                             ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c: In function 'cfg80211_rtw_add_key':
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:1418:35: warning: passing argument 2 of '_rtw_memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   _rtw_memcpy(param->u.crypt.seq, params->seq, params->seq_len);
                                   ^
In file included from /home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:23:0:
/home/alarm/rtl8192/rtl8192cu-fixes/include/osdep_service.h:1398:13: note: expected 'void *' but argument is of type 'const u8 * {aka const unsigned char *}'
 extern void _rtw_memcpy(void* dec, void* sour, u32 sz);
             ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:1424:35: warning: passing argument 2 of '_rtw_memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   _rtw_memcpy(param->u.crypt.key, params->key, params->key_len);
                                   ^
In file included from /home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:23:0:
/home/alarm/rtl8192/rtl8192cu-fixes/include/osdep_service.h:1398:13: note: expected 'void *' but argument is of type 'const u8 * {aka const unsigned char *}'
 extern void _rtw_memcpy(void* dec, void* sour, u32 sz);
             ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c: In function 'cfg80211_rtw_get_station':
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:1592:20: error: 'STATION_INFO_SIGNAL' undeclared (first use in this function)
   sinfo->filled |= STATION_INFO_SIGNAL;
                    ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:1595:20: error: 'STATION_INFO_TX_BITRATE' undeclared (first use in this function)
   sinfo->filled |= STATION_INFO_TX_BITRATE;
                    ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:1598:20: error: 'STATION_INFO_RX_PACKETS' undeclared (first use in this function)
   sinfo->filled |= STATION_INFO_RX_PACKETS;
                    ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:1601:20: error: 'STATION_INFO_TX_PACKETS' undeclared (first use in this function)
   sinfo->filled |= STATION_INFO_TX_PACKETS;
                    ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c: In function 'cfg80211_rtw_connect':
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:2728:30: warning: passing argument 2 of '_rtw_memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  _rtw_memcpy(ndis_ssid.Ssid, sme->ssid, sme->ssid_len);
                              ^
In file included from /home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:23:0:
/home/alarm/rtl8192/rtl8192cu-fixes/include/osdep_service.h:1398:13: note: expected 'void *' but argument is of type 'const u8 * {aka const unsigned char *}'
 extern void _rtw_memcpy(void* dec, void* sour, u32 sz);
             ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:2763:42: warning: passing argument 2 of 'rtw_cfg80211_set_wpa_ie' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  ret = rtw_cfg80211_set_wpa_ie(padapter, sme->ie, sme->ie_len);
                                          ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:2422:12: note: expected 'u8 * {aka unsigned char *}' but argument is of type 'const u8 * {aka const unsigned char *}'
 static int rtw_cfg80211_set_wpa_ie(_adapter *padapter, u8 *pie, size_t ielen)
            ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:2850:39: warning: passing argument 2 of 'rtw_set_802_11_connect' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  if (rtw_set_802_11_connect(padapter, sme->bssid, &ndis_ssid) == _FALSE) {
                                       ^
In file included from /home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:26:0:
/home/alarm/rtl8192/rtl8192cu-fixes/include/rtw_ioctl_set.h:66:4: note: expected 'u8 * {aka unsigned char *}' but argument is of type 'const u8 * {aka const unsigned char *}'
 u8 rtw_set_802_11_connect(_adapter* padapter, u8 *bssid, NDIS_802_11_SSID *ssid);
    ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c: In function 'cfg80211_rtw_set_pmksa':
[... 56 lines cut ...]
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:5177:17: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .add_station = cfg80211_rtw_add_station,
                 ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:5177:17: note: (near initialization for 'rtw_cfg80211_ops.add_station')
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:5178:17: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .del_station = cfg80211_rtw_del_station,
                 ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:5178:17: note: (near initialization for 'rtw_cfg80211_ops.del_station')
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:5179:20: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .change_station = cfg80211_rtw_change_station,
                    ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:5179:20: note: (near initialization for 'rtw_cfg80211_ops.change_station')
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:5195:13: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .mgmt_tx = cfg80211_rtw_mgmt_tx,
             ^
/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.c:5195:13: note: (near initialization for 'rtw_cfg80211_ops.mgmt_tx')
scripts/Makefile.build:258: recipe for target '/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/home/alarm/rtl8192/rtl8192cu-fixes/os_dep/linux/ioctl_cfg80211.o] Error 1
Makefile:1384: recipe for target '_module_/home/alarm/rtl8192/rtl8192cu-fixes' failed
make[1]: *** [_module_/home/alarm/rtl8192/rtl8192cu-fixes] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.1.18-1-ARCH/build'
Makefile:596: recipe for target 'modules' failed
make: *** [modules] Error 2

Need to fix for support for nl80211 driver, and allow Wifi-Direct Interface.

pvaret, 2016-02-24:
This looks like a bug in the upstream Realtek driver. Sadly, they have not
historically been great about acting on bug reports.

I understand there is a new native 8192CU kernel driver coming in kernel
4.5. That is, a driver redeveloped from the ground up to properly make use
of the WiFi driver infrastructure of the kernel. (This here, essentially
Realtek's own out-of-tree driver, just made easier to install, reinvents
pretty much every wheel on the way, badly.) Have you tried that one? Even
if it's not quite there yet, the kernel developers are bound to be more
responsive to bug reports than Realtek.

bass0324, 2016-02-24:
Yea definitely an issue with the driver source.  I was attempting to build for arch Linux and not raspbian, as I know for a fact that the 'firmware-realtek' package in raspbian allows for wifi direct communication, but not sure it allows for wireless display, which also needs to be enabled during compile. This is why I wanted to build from source rather than rely on pre-built packages. 

The other alternative would be to downgrade the kernel in order to take advantage of the source.

kyawthusoe314, 2016-02-27:
Dear sir pvaret, the mighty rtl8192cu-fixes of yours no longer compile on fedora 23. It did on Debian Jessie and Ubuntu 14.04 LTS. How can we solve it? :(

pvaret, 2016-03-01:
Hi @kyawthusoe314,

Right, let me try to unpile everything here.

- You posted in an issue report about RPi, but mentioned Fedora 23. Which is it? I'm guessing Fedora? It would be better to open a new issue, since this is a separate one.

- This repository is for Debian and Ubuntu. We've been told it works with Fedora with some fiddling, which is cool, but that's only hearsay. I don't use Fedora and I'm afraid I can't help Fedora users for any Fedora-specific issue.

- There's no chance we'll be able to help if you don't even tell us what went wrong in the first place!

Consider trying out kernel 4.4? It comes with a brand new 8192CU driver rewritten from scratch.

stefson, 2016-03-02:
If you would like to have a working kernel modul for Rpi, you should try the 8192cu-dkms package from aur. It is mandatory to disable power management though.

bass0324, 2016-03-02:
The issue is not a just a working kernel module, but a working kernel module with wifi direct support. When attempting any iwpriv command with default module it fails to find private ioctls. Then, in order to enable wifi direct support, nl80211 support is also needed, which I believe wasn't updated to support the new kernel commands introduced in kernels >3.10. This fixed repository has updated the main files which allow the kernel module to build under the latest kernels, but did not update the files needed to enable nl80211 support.

kyawthusoe314, 2016-03-08:
Quite responsive! It's either my mistake or the fedora 23. Besides, 23 won't hibernate and that's another issues. I'd stick with Debian meanwhile.:)

pvaret, 2017-10-14:
As a late update: the rtl8xxxu driver from the mainline kernel works pretty fine these days. That's what I'm using now myself. You may still have to troubleshoot NetworkManager, though: in particular, consider [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Disabling it made the rtlxxxu driver work well for me.

closed 2017-10-14 by pvaret