pvaret/rtl8192cu-fixes #70: Issues with Debian Stretch x64

speendo, 2016-03-17:
I had this driver on my desktop with Debian Stretch x64 working for several months now.

Since last weekend, it stopped working well. When I plug in my Edimax EW7811-UN udev states
```
[ 3748.960955] usb 4-1.8: new high-speed USB device number 4 using ehci-pci
[ 3749.054686] usb 4-1.8: New USB device found, idVendor=7392, idProduct=7811
[ 3749.054691] usb 4-1.8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3749.054694] usb 4-1.8: Product: 802.11n WLAN Adapter
[ 3749.054696] usb 4-1.8: Manufacturer: Realtek
[ 3749.054698] usb 4-1.8: SerialNumber: 00e04c000001
[ 3750.234522] rtl8192cu 4-1.8:1.0 wlx74da38006385: renamed from wlan0
```
What does the last line mean? wlan0 is a good name in my opinion.

Anyway, this wouldn't be a problem, but the wifi dongle is also not listed in my Gnome 3 network menu.

I think this might be related...

I'm glad for any help...

pvaret, 2016-03-18:
Device naming is udev's business, not the driver's. Look here for the relevant documentation: https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/

closed 2016-03-18 by pvaret

speendo, 2016-03-19:
Thanks for looking at it.

I'm not saying that the renaming is the problem here. I thought that first but in the mean time I'm convinced that it's not the case.

Still I experience a change in the behavior of my dongle. Quite a big one: until last week it worked flawlessly (and was listed in Gnome 3 NetworkManager) and now it doesn't work anymore.

Using the default driver (rtl8192cu) the dongle still is listed in the NetworkManager as expected. Reinstalling the driver from this repo it isn't.

Therefore I would think that the issue is related with some incompatibility between Debian Stretch x64 packages and this project. However, I cannot be sure yet.

Can I provide any additional information to clear things up?

Thank you in advance!

speendo, 2016-03-19:
Ok, just made another system update. Now it seems to work again. At least for the moment. I hope it stays like this...

Thank you for your help!