pvaret/rtl8192cu-fixes #71: Problem with armv5 based system

wimpunk, 2016-03-30:
I'm trying to get this driver working on armv5 (actually a Freescale i.MX287 based karo tx28) but I seem to miss something.  The new driver loads correctly while inserting the USB wifi and ifconfig shows me a device:
```
$ ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr 6c:19:8f:b8:ae:06
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

Unfortunately, every `iw` command seems to fail although they seems to work on raspberry pi 2.  When I try run wpa_supplicant with the same arguments as the kernel driver, I get this:

```
$ /sbin/wpa_supplicant -s -B -P /run/wpa_supplicant.wlan0.pid -i wlan0 -D nl80211,wext -c /etc/wpa_supplicant/wpa_supplicant.conf
ioctl[SIOCSIWMODE]: Operation not supported
ioctl[SIOCGIWRANGE]: Operation not supported
ioctl[SIOCGIWMODE]: Operation not supported
ioctl[SIOCSIWAP]: Operation not supported
ioctl[SIOCSIWESSID]: Operation not supported
ioctl[SIOCSIWENCODEEXT]: Operation not supported
ioctl[SIOCSIWENCODE]: Operation not supported
ioctl[SIOCSIWENCODEEXT]: Operation not supported
ioctl[SIOCSIWENCODE]: Operation not supported
ioctl[SIOCSIWENCODEEXT]: Operation not supported
ioctl[SIOCSIWENCODE]: Operation not supported
ioctl[SIOCSIWENCODEEXT]: Operation not supported
ioctl[SIOCSIWENCODE]: Operation not supported
ioctl[SIOCSIWENCODEEXT]: Operation not supported
ioctl[SIOCSIWENCODE]: Operation not supported
ioctl[SIOCSIWENCODEEXT]: Operation not supported
ioctl[SIOCSIWENCODE]: Operation not supported
```

Is there anyone out there who can help me?  I'm running a 3.18.28-based kernel.

allaudet, 2016-03-30:
If I'm not wrong, this driver by default only supports **wext** and seems like your Kernel does not support it.

The linux command `$ iw` is used to configure devices under the **cfg80211** standard. So it won't work with this driver. You have to use `$ iwconfig` instead.

Be aware that you will need to use a specific Hostapd modified version if you want to configure your device as Access Point.

Under what OS are you working?
If you are building the Kernel, you need to add **`DRIVER_WEXT_SUPPORT`** as a kernel config.

wimpunk, 2016-04-07:
Sorry, I missed your reply somehow.  I'll check it asap and let you know if your suggestions work.

wimpunk, 2016-04-07:
`WEXT` seems to be supported but not by `DRIVER_WEXT_SUPPORT`  I'm getting this:
```
# zgrep WEXT /proc/config.gz
CONFIG_WEXT_CORE=y
CONFIG_WEXT_PROC=y
CONFIG_CFG80211_WEXT=y
```
Grepping for `DRIVER_WEXT_SUPPORT` doesn't give me any result in `.config`.

I'm working on a karo TX28 with a homebrew kernel.  The linux system (except for the kernel and modules) is a debian 8.3.

dlech, 2016-04-07:
Here is my working configuration from an armv5 system: http://paste.ubuntu.com/15672143/

It a 3.16 kernel, so should be close enough. Hope it helps.

wimpunk, 2016-04-08:
Thanks for the configuration, @dlech.
It looks like the `DRIVER_WEXT_SUPPORT` is activated while activating some drivers.  I've tried to force to enable it but it currently doesn't seem to work.  I'm still getting the same result but I'll try it again later on.

pvaret, 2017-10-14:
As a late update: the rtl8xxxu driver from the mainline kernel works pretty fine these days. That's what I'm using now myself. You may still have to troubleshoot NetworkManager, though: in particular, consider [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Disabling it made the rtlxxxu driver work well for me.

closed 2017-10-14 by pvaret