pvaret/rtl8192cu-fixes #74: Problems with new version of NetworkManager

mfxbe, 2016-04-09:
After an update from _NetworkManager-1.0.6_ to _NetworkManager-1.0.10_ the driver stops working. It seems for me that the device isn't recognized the correct way. At the moment I just tried it on Fedora 23, so also could be distro specific.

I'm not really  familiar with the drivers, so I don't no what further informations would be useful. Of course I could provide some (just ask).

PS: If someone else have the problem too, just downgrade NetworkManager works for the moment.

GryfonN, 2016-04-10:
same problem on my Fedora23

`
uname -r 
4.4.5-300.fc23.x86_64
`

fat-lobyte, 2016-04-16:
Hi, I have the same problem, also on Fedora 23.

Kernel: 4.4.6-301.fc23.x86_64
NetworkManager: 1.0.12-1.fc23.x86_64

If you need more information, I am welcome to provide it.

pvaret, 2016-04-16:
Thanks for the heads-up. I'm still on NM 1.0.4 here; I'll keep an eye out when the next Ubuntu rolls out.

nv95, 2016-04-19:
Fixed with latest version of NetworkManager on Fedora 23

GryfonN, 2016-04-19:
Yes it works on Fedora 23, kernel 4.4.6-301.fc23.x86_64, NetworkManager 1.0.12-2.fc23

mfxbe, 2016-04-30:
I can confirm, seems to work again.

closed 2016-04-30 by mfxbe