pvaret/rtl8192cu-fixes #75: doesn' t work  on linux 4.4.0 (ubuntu 16.04)

mapelo, 2016-04-16:
doesn' t work  on linux 4.4.0 (ubuntu 16.04)

Solution: You need to add rtl8xxxu into blacklist-native-rtl8192.conf

pvaret, 2016-04-16:
Thanks for the heads-up.

Isn't kernel 4.4 supposed to fix its own rtl8192cu driver, making this one obsolete, though? (Haven't tested it yet.)

kyawthusoe3142, 2016-04-21:
@mapelo How to add rtl8xxxu into blacklist-native-rtl8192.conf?

dlech, 2016-04-22:
See <https://github.com/pvaret/rtl8192cu-fixes/blob/master/blacklist-native-rtl8192.conf>

Add the line:

    install rtl8xxxu /bin/false

kyawthusoe3142, 2016-04-24:
Thank you! It worked.

dlech, 2016-04-26:
Fixed: https://github.com/pvaret/rtl8192cu-fixes/commit/fcb0eccb68509c3d64ac9f35a3f1ab3ff83cf5a9

closed 2016-04-26 by dlech

otakuryo, 2016-09-23:
2 weeks and you save me, I love you *-*
(Arch 4.7.4)

dcdeiv, 2016-10-04:
Ehrm no. Xubuntu 16.04 LTS it doesn't work. Wifi now doesn't exists anymore. I had to reinstall Xubuntu.

ajith4ab, 2017-12-27:
@mapelo cannot find rtl8xxxu into blacklist-native-rtl8192.conf?