pvaret/rtl8192cu-fixes #76: Low signal on Ubuntu 16.04

Flix01, 2016-04-25:
Hi everybody.
I've been using this driver happily for many months (almost one year I guess): thank you very much for your work!

On my update to Ubuntu 16.04 it still works, but I keep seeing the "wifi meter" at the top of the screen at 2/4 (half the speed).

I'm not sure if this is a temporary issue or it is related with the driver/kernel pair, after my upgrade to Ubuntu 16.04.

This is my config:
```
> dkms status 8192cu | grep installed
8192cu, 1.10, 3.19.0-31-generic, x86_64: installed
8192cu, 1.10, 4.2.0-35-generic, x86_64: installed
8192cu, 1.10, 4.4.0-21-generic, x86_64: installed

> lsusb
Bus 003 Device 002: ID 0b05:17ba ASUSTek Computer, Inc. N10 Nano 802.11n Network Adapter [Realtek RTL8192CU]

> iwconfig
wlan0     IEEE 802.11bgn  ESSID:"..."  
          Mode:Managed  Frequency:2.437 GHz  Access Point: ...   
          Bit Rate=1 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Power Management:off
          Link Quality=36/70  Signal level=-74 dBm  		<== TOO LOW!
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:549   Missed beacon:0
```

I'd like to stay on the "safe side" (= I prefer a low connection to a non-working one): so I'm not sure whether to try something or not...

Any advice (please speak to me like a child: I know almost nothing about this topic...).

P.S. Does somebody know if Ubuntu 16.04 has a working driver for these devices ? Why rtl8192cu-fixes
is not "embedded" in it ?

Flix01, 2016-04-25:
Hey! Wait a minute! Am I really using <b>your</b> driver ?

When I click on <b>connection information</b> I get: <b>Driver: rtl8xxxu</b>
And if I type in a console:
```
> sudo lshw -C network
  *-network
       description: Wireless interface
       physical id: 1
       bus info: usb@3:2
       logical name: wlan0
       serial: ...
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8xxxu driverversion=4.4.0-21-generic firmware=N/A ip=... link=yes multicast=yes wireless=IEEE 802.11bgn

> find /lib/modules/$(uname -r)/kernel/drivers/net/wireless -name '*.ko'
[...]
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtlwifi.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192ee/rtl8192ee.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192de/rtl8192de.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl_usb.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192cu/rtl8192cu.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192se/rtl8192se.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8723com/rtl8723-common.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192c/rtl8192c-common.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8188ee/rtl8188ee.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8723ae/rtl8723ae.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8821ae/rtl8821ae.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8723be/rtl8723be.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/btcoexist/btcoexist.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192ce/rtl8192ce.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl_pci.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtl818x/rtl8187/rtl8187.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtl818x/rtl8180/rtl818x_pci.ko
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtl8xxxu/rtl8xxxu.ko
[...]
```

As you can see there is a:
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtl8xxxu/<b>rtl8xxxu.ko</b>
that is running.

Your driver is probably:
/lib/modules/4.4.0-21-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192cu/<b>rtl8192cu.ko</b>

Is this correct? How can I re-enable it ?

P.S. I thought I had followed all the steps in your Github home page to install it...
Am I missing something ?

dlech, 2016-04-25:
> Is this correct? How can I re-enable it ?

Yes. There is a new driver (rtl8xxxu) in the Linux kernel since 4.? See #75 on how to blacklist the new driver.

Flix01, 2016-04-25:
@dlech Thanks. I'll try it soon!

Flix01, 2016-04-25:
It works! And it makes a <b>HUGE</b> difference!

Now <b>Connection Information</b> states: <b>Driver: rtl8192cu</b> 

But, furthermore:
```
> iwconfig
wlan0     IEEE 802.11bgn  ESSID:"..."  Nickname:"<...>"
          Mode:Managed  Frequency:2.437 GHz  Access Point: ...  
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=97/100  Signal level=60/100  Noise level=0/100    <= SEE ANY DIFFERENCE ?
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

@dlech Thank you very, very, very much.

P.S. I think https://github.com/pvaret/rtl8192cu-fixes/blob/master/blacklist-native-rtl8192.conf should be updated to include: install rtl8xxxu /bin/false.
Otherwise many people will incorrectly think they're still using this driver...

dlech, 2016-04-26:
done: https://github.com/pvaret/rtl8192cu-fixes/commit/fcb0eccb68509c3d64ac9f35a3f1ab3ff83cf5a9

closed 2016-04-26 by dlech

pvaret, 2016-04-26:
Thanks, David. :)​