pvaret/rtl8192cu-fixes #77: ubuntu 16.04

Ryogaag, 2016-04-26:
I installed fresh ubuntu 16.04 and tried this method, when i try to connect to my wifi it doesn't work and keep asking for the password of the network

pvaret, 2016-04-26:
Can you try updating to the very latest version (fcb0ecc) and rebooting? Ubuntu recently added a new driver that wasn't in our blacklist until half a day ago.

Ryogaag, 2016-04-26:
After the third restart it worked, but it's very slow.
I already checked for the blacklist-native-rtl8192.conf and it's correct.
What can i do?

pvaret, 2016-04-26:
Did you have this problem before switching to Ubuntu 16.04?

dlech, 2016-04-26:
Also, what is output of `lsmod`?

Ryogaag, 2016-04-27:
before it worked fine.
This is the lsmod result:

> Module                  Size  Used by
> snd_usb_audio         176128  0
> snd_usbmidi_lib        36864  1 snd_usb_audio
> nls_iso8859_1          16384  1
> 8192cu                532480  0
> mac80211              737280  0
> cfg80211              565248  1 mac80211
> gspca_ov534            20480  0
> gspca_main             36864  1 gspca_ov534
> v4l2_common            16384  1 gspca_main
> videodev              176128  3 gspca_main,v4l2_common,gspca_ov534
> media                  24576  1 videodev
> eeepc_wmi              16384  0
> asus_wmi               28672  1 eeepc_wmi
> sparse_keymap          16384  1 asus_wmi
> video                  40960  1 asus_wmi
> kvm_amd                65536  0
> kvm                   536576  1 kvm_amd
> snd_hda_codec_realtek    81920  1
> snd_hda_codec_generic    77824  1 snd_hda_codec_realtek
> snd_hda_codec_hdmi     53248  1
> snd_hda_intel          36864  5
> snd_hda_codec         135168  4 snd_hda_codec_realtek,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_intel
> snd_hda_core           73728  5 snd_hda_codec_realtek,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_codec,snd_hda_intel
> snd_hwdep              16384  2 snd_usb_audio,snd_hda_codec
> joydev                 20480  0
> input_leds             16384  0
> snd_pcm               106496  5 snd_usb_audio,snd_hda_codec_hdmi,snd_hda_codec,snd_hda_intel,snd_hda_core
> irqbypass              16384  1 kvm
> snd_seq_midi           16384  0
> snd_seq_midi_event     16384  1 snd_seq_midi
> snd_rawmidi            32768  2 snd_usbmidi_lib,snd_seq_midi
> snd_seq                69632  2 snd_seq_midi_event,snd_seq_midi
> snd_seq_device         16384  3 snd_seq,snd_rawmidi,snd_seq_midi
> crct10dif_pclmul       16384  0
> crc32_pclmul           16384  0
> snd_timer              32768  2 snd_pcm,snd_seq
> aesni_intel           167936  0
> aes_x86_64             20480  1 aesni_intel
> lrw                    16384  1 aesni_intel
> gf128mul               16384  1 lrw
> glue_helper            16384  1 aesni_intel
> ablk_helper            16384  1 aesni_intel
> snd                    81920  23 snd_hda_codec_realtek,snd_usb_audio,snd_hwdep,snd_timer,snd_hda_codec_hdmi,snd_pcm,snd_seq,snd_rawmidi,snd_hda_codec_generic,snd_usbmidi_lib,snd_hda_codec,snd_hda_intel,snd_seq_device
> soundcore              16384  1 snd
> serio_raw              16384  0
> cryptd                 20480  2 aesni_intel,ablk_helper
> fam15h_power           16384  0
> k10temp                16384  0
> edac_mce_amd           24576  0
> edac_core              53248  0
> i2c_piix4              24576  0
> shpchp                 36864  0
> mac_hid                16384  0
> 8250_fintek            16384  0
> tpm_infineon           20480  0
> parport_pc             32768  0
> ppdev                  20480  0
> lp                     20480  0
> parport                49152  3 lp,ppdev,parport_pc
> autofs4                40960  2
> uas                    24576  0
> usb_storage            69632  2 uas
> hid_generic            16384  0
> usbhid                 49152  0
> hid                   118784  2 hid_generic,usbhid
> amdkfd                122880  1
> amd_iommu_v2           20480  1 amdkfd
> radeon               1511424  3
> mxm_wmi                16384  0
> i2c_algo_bit           16384  1 radeon
> ttm                    98304  1 radeon
> drm_kms_helper        139264  1 radeon
> psmouse               126976  0
> syscopyarea            16384  1 drm_kms_helper
> sysfillrect            16384  1 drm_kms_helper
> sysimgblt              16384  1 drm_kms_helper
> r8169                  81920  0
> fb_sys_fops            16384  1 drm_kms_helper
> mii                    16384  1 r8169
> drm                   360448  45 ttm,drm_kms_helper,radeon
> ahci                   36864  3
> libahci                32768  1 ahci
> fjes                   28672  0
> wmi                    20480  2 mxm_wmi,asus_wmi
>

pvaret, 2016-04-27:
Hmm. The odd thing is, this driver did not change, so your issues seem to indicate that something *else* did. It's a long shot, but could you try the following?

    sudo modprobe -r mac80211
    sudo modprobe -r cfg80211

Ryogaag, 2016-04-27:
It's still unstable
here's the lsmod

> Module                  Size  Used by
> 8192cu                532480  0
> snd_usb_audio         176128  0
> snd_usbmidi_lib        36864  1 snd_usb_audio
> nls_iso8859_1          16384  1
> mac80211              737280  0
> cfg80211              565248  1 mac80211
> gspca_ov534            20480  0
> gspca_main             36864  1 gspca_ov534
> v4l2_common            16384  1 gspca_main
> videodev              176128  3 gspca_main,v4l2_common,gspca_ov534
> media                  24576  1 videodev
> eeepc_wmi              16384  0
> asus_wmi               28672  1 eeepc_wmi
> sparse_keymap          16384  1 asus_wmi
> video                  40960  1 asus_wmi
> kvm_amd                65536  0
> kvm                   536576  1 kvm_amd
> input_leds             16384  0
> joydev                 20480  0
> snd_hda_codec_realtek    81920  1
> snd_hda_codec_generic    77824  1 snd_hda_codec_realtek
> snd_hda_codec_hdmi     53248  1
> irqbypass              16384  1 kvm
> snd_hda_intel          36864  5
> snd_hda_codec         135168  4 snd_hda_codec_realtek,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_intel
> snd_hda_core           73728  5 snd_hda_codec_realtek,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_codec,snd_hda_intel
> crct10dif_pclmul       16384  0
> snd_hwdep              16384  2 snd_usb_audio,snd_hda_codec
> crc32_pclmul           16384  0
> snd_pcm               106496  5 snd_usb_audio,snd_hda_codec_hdmi,snd_hda_codec,snd_hda_intel,snd_hda_core
> snd_seq_midi           16384  0
> snd_seq_midi_event     16384  1 snd_seq_midi
> snd_rawmidi            32768  2 snd_usbmidi_lib,snd_seq_midi
> aesni_intel           167936  0
> snd_seq                69632  2 snd_seq_midi_event,snd_seq_midi
> snd_seq_device         16384  3 snd_seq,snd_rawmidi,snd_seq_midi
> aes_x86_64             20480  1 aesni_intel
> snd_timer              32768  2 snd_pcm,snd_seq
> lrw                    16384  1 aesni_intel
> snd                    81920  23 snd_hda_codec_realtek,snd_usb_audio,snd_hwdep,snd_timer,snd_hda_codec_hdmi,snd_pcm,snd_seq,snd_rawmidi,snd_hda_codec_generic,snd_usbmidi_lib,snd_hda_codec,snd_hda_intel,snd_seq_device
> gf128mul               16384  1 lrw
> soundcore              16384  1 snd
> glue_helper            16384  1 aesni_intel
> shpchp                 36864  0
> ablk_helper            16384  1 aesni_intel
> serio_raw              16384  0
> cryptd                 20480  2 aesni_intel,ablk_helper
> edac_mce_amd           24576  0
> edac_core              53248  0
> fam15h_power           16384  0
> k10temp                16384  0
> i2c_piix4              24576  0
> mac_hid                16384  0
> 8250_fintek            16384  0
> tpm_infineon           20480  0
> parport_pc             32768  0
> ppdev                  20480  0
> lp                     20480  0
> parport                49152  3 lp,ppdev,parport_pc
> autofs4                40960  2
> uas                    24576  0
> usb_storage            69632  2 uas
> hid_generic            16384  0
> usbhid                 49152  0
> hid                   118784  2 hid_generic,usbhid
> amdkfd                122880  1
> amd_iommu_v2           20480  1 amdkfd
> radeon               1511424  3
> mxm_wmi                16384  0
> i2c_algo_bit           16384  1 radeon
> ttm                    98304  1 radeon
> drm_kms_helper        139264  1 radeon
> psmouse               126976  0
> syscopyarea            16384  1 drm_kms_helper
> sysfillrect            16384  1 drm_kms_helper
> sysimgblt              16384  1 drm_kms_helper
> r8169                  81920  0
> fb_sys_fops            16384  1 drm_kms_helper
> mii                    16384  1 r8169
> drm                   360448  49 ttm,drm_kms_helper,radeon
> ahci                   36864  3
> libahci                32768  1 ahci
> fjes                   28672  0
> wmi                    20480  2 mxm_wmi,asus_wmi
> 

Thanks for the help you're giving to me :)

speendo, 2016-05-03:
I seem to have similar issues on Debian Stretch

ChadFulton, 2016-05-19:
I have no idea what I'm doing here, but I just did clean installs of Ubuntu 16.04 on two machines with the Edimax EW-7811Un chip. As usual, wireless was terrible out of the box.

Both times when I ran through the instructions on the site once and rebooted, it did not recognize that there was a wifi device available. If I removed the blacklist file, it recognized the native driver and had bad performance again.

However, both times, when I just ran through the same instructions a second time, starting at "Set it up as a DKMS module:" and continuing through re-adding the blacklist, it then worked perfectly after reboot.

I'm almost positive that I didn't miss any of the steps the first time through, and the exact same thing did happen this same way with two separate installations.

By the way, thanks for a great fix.

dbrons, 2016-05-24:
Same for me on Ubuntu 15.10 , kernel 4.2.0.-36 ...

output lsusb **BUS 001 Device 002: ID 0ba:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter**

output lsmod, as expected, these modules was loaded : **mac80211 + cfg80211**.

8192cu is loaded, but not in use.

modprobe -r mac80211
modprobe -r cfg80211

done, but didn't solve the problem.

I added these lines in /etc/modprobe.d/blacklist-native-rtl8192.conf

**install mac80211 /bin/false
install cfg80211 /bin/false**

after a reboot, they are not anymore loaded in **lsmod**

output of ethtool -i wlex8de2709c6c0:
**driver: rtl8192cu** 

...

dbrons, 2016-05-24:
I forget to say that I am working on ubuntu under virtualbox

I added these lines: 

**echo '8192cu' | sudo tee -a /etc/modules
echo "0bda 8178" | sudo tee /sys/bus/usb/drivers/rtl8192cu/new_id**

+ at etc/rc.local 

**echo "0bda 8178" | tee /sys/bus/usb/drivers/rtl8192cu/new_id
exit 0**

output of sudo lshw -C network
**sudo lshw -C network
  *-network               
       description: Wireless interface
       physical id: 1
       bus info: usb@1:1
       logical name: wlxe8de2709c6c0
       serial: e8:de:27:09:c6:c0
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192cu multicast=yes wireless=unassociated**

output of dmesg | grep 8192
**[  359.310494] 8192cu: module verification failed: signature and/or required key missing - tainting kernel
[  361.743969] usbcore: registered new interface driver rtl8192cu
[  363.084621] rtl8192cu 1-1:1.0 wlxe8de2709c6c0: renamed from wlan0**

output of lsmod | grep 8192cu
**8192cu                532480  0**

dbrons, 2016-05-31:
just for inform you that under a freshinstall of Ubuntu 16.04 on a laptop, I have nor problem with kernel driver rtl8192cu ...

ashishkotnala29, 2016-07-09:
@dbrons Do you mean that the default built-in kernel driver rtl8192cu works in kernel 4.4+, therefore no need to build this driver anymore?
I also have a Netgear WNA1000M USB WiFi adapter with rtl8188csu chipset which uses same kernel driver, so in Ubuntu 16.04 it's fully supported?

dbrons, 2016-07-10:
Hey @ashishkotnala29 !  Indeed with the built-in driver from Ubuntu 16.04, I had no problem with my wifi dongle (TL-WN822N).  All my problems was coming from a use of the 8192cu fixe (or rtl8192cu built-in driver)  on any verison of Ubuntu under VirtualBox.

dcdeiv, 2016-10-04:
It does not see any wifi now after installing the fix

roboticboyer, 2017-06-11:
I have fixed this issue per [comment](https://github.com/pvaret/rtl8192cu-fixes/issues/77#issuecomment-220238555)

pvaret, 2017-10-14:
As a late update: I found that issues with intermittently unstable connections seemed to be caused by power management issues in NetworkManager. In short, consider [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Disabling it made the driver work well for me. You may also consider switching to the rtl8xxxu driver from the mainline kernel, which works well these days.

closed 2017-10-14 by pvaret