pvaret/rtl8192cu-fixes #80: AP mode not supported ?

sergiozia, 2016-05-09:
Hi,

I'am under Linux Mint 17.3 and encountered some problems with my wi-fi hotspot based on a TP-Link TL-WN823N USB dongle using the Realtek RTL8192CU chipset.
The hotspot created with NetworkManager was fully operational but client connection was frequently dropped and not recoverable until the hostpot was restarted.

Thus I installed your driver hoping that it will solve the problem.
But now it is no longer possible to activate my hotspot !

This message appears into the /var/log/syslog :

NetworkManager[1114]: <warn> connection /org/freedesktop/NetworkManager/Settings/24 failed to activate: (5) Access Point (AP) mode is not supported by this device.

Here is my NetworkManager Hotspot config : 

[connection]
id=Hotspot
uuid=c09be793-757d-40f0-a85f-9d966e297f59
type=802-11-wireless
autoconnect=false

[802-11-wireless]
ssid=ipmwks4298
mode=ap
mac-address=F4:F2:6D:1B:2D:A6
security=802-11-wireless-security

[802-11-wireless-security]
key-mgmt=wpa-psk
proto=rsn;
pairwise=ccmp;
group=ccmp;
psk=MY_Hotspot!

[ipv4]
method=shared

[ipv6]
method=auto

allaudet, 2016-05-09:
> type=802-11-wireless

As far as I know, this driver does not support **mac80211**, only WEXT.

You should use _hostapd_ with the **rtl871xdrv** driver patch compiled.

sergiozia, 2016-05-09:
Thanks for you reply but... how to install hostapd with the rtl871xdrv driver patch compiled ? Do you have a procedure ?
I'm sorry but I'm not familiar with network interfaces.
Also, after that, how to configure my hotspot using hostapd ?

dlech, 2016-05-09:
Duplicate of #32. See my [response there](https://github.com/pvaret/rtl8192cu-fixes/issues/32#issuecomment-217896120).

closed 2016-05-09 by dlech