pvaret/rtl8192cu-fixes #86: os_intfs.c:1003:22: error: initialization from incompatible pointer type

marcosscriven, 2016-05-20:
Trying to build on BeagleBone Black:
```
ubuntu@arm:~$ uname -a
Linux arm 4.4.9-ti-r25 #1 SMP Fri May 6 10:28:56 UTC 2016 armv7l armv7l armv7l GNU/Linux
```

It's failing with:

```
  CC [M]  /var/lib/dkms/8192cu/1.10/build/os_dep/linux/os_intfs.o
/var/lib/dkms/8192cu/1.10/build/os_dep/linux/os_intfs.c:1003:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^
/var/lib/dkms/8192cu/1.10/build/os_dep/linux/os_intfs.c:1003:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
scripts/Makefile.build:258: recipe for target '/var/lib/dkms/8192cu/1.10/build/os_dep/linux/os_intfs.o' failed
make[1]: *** [/var/lib/dkms/8192cu/1.10/build/os_dep/linux/os_intfs.o] Error 1
Makefile:1387: recipe for target '_module_/var/lib/dkms/8192cu/1.10/build' failed
make: *** [_module_/var/lib/dkms/8192cu/1.10/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.4.9-ti-r25'
```

dlech, 2016-05-20:
Apparently most people must be compiling this without `LINUX_VERSION_CODE` set because the function signatures don't match. See [here](https://github.com/pvaret/rtl8192cu-fixes/blob/master/os_dep/linux/os_intfs.c#L1002).

You could copy the function signature from [here](https://github.com/raspberrypi/linux/blob/3b1047181fbbbd2067b6b7476c428199474fdd19/drivers/staging/rtl8723au/os_dep/os_intfs.c#L296) and replace the one [here](https://github.com/pvaret/rtl8192cu-fixes/blob/master/os_dep/linux/os_intfs.c#L952) and it should work.

Looks like the [breaking upstream change](https://github.com/torvalds/linux/commit/99932d4fc03a13bb3e94938fe25458fabc8f2fc3) has been around since kernel 3.14

dlech, 2016-05-20:
Check out the `rtw_select_queue` branch. If it compiles and works, I will merge #87.

marcosscriven, 2016-05-22:
👍 thanks - I was just able to build and install from that branch:
```
ubuntu@arm:~$ lsmod | grep 8192
8192cu                429353  0
```

Unfortunately though I'm still getting connection issues, but I guess it's some other configuration issue.

It's interesting your linked to the RPi kernel repo - the same Edimax wifi dongle works fine with Raspbian - are some/most of your changes merged into the RPi kernel? Have you ever tried to get them into the mainline kernel?

dlech, 2016-05-22:
The RPi kernel uses the same source code from Realtek that this repository is based on. I imagine they made essentially the same changes that have been made here.

Yes [we have tried unsuccessfully to get this mainlined](https://github.com/pvaret/rtl8192cu-fixes/issues/29). It is not acceptable because it re-implements parts of the networking stack instead of using the existing kernel interfaces.

There is a new compatible driver in the mainline kernel since v4.4. However, it doesn't have very good performance compared to this driver, which is [why RPi is still shipping a version of this driver](https://github.com/raspberrypi/linux/pull/1468). You can see the full history of RPi's changes [here](https://github.com/raspberrypi/linux/commits/rpi-4.4.y/drivers/net/wireless/realtek/rtl8192cu).

It is interesting that the RPi kernel does not have the change that I proposed here, so apparently, the affected code is not being compiled in the RPi kernel. Apparently, you are compiling with different compiler options enabled or disabled.

marcosscriven, 2016-05-23:
@pvaret - thanks so much for taking the time to respond so comprehensively; understanding what changes are in which kernels (and where) is useful to know. Thanks also for fixing the compilation issue so quickly.

Unfortunately I'm still seeing issues on the BBB, but I think it must be down to power issues. For anyone looking at this in the future, I've followed on trying to get this work with this post to the BeagleBone Google group: https://groups.google.com/forum/#!category-topic/beagleboard/beaglebone-black/a4v9-xgQitE

pvaret, 2017-10-14:
For power-related issues, consider just [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Disabling it made the driver work well for me.

PR #87 (2016-05-22): Add #if for 3.14 kernel change
Fixes compiling after https://github.com/torvalds/linux/commit/99932d4fc03a13bb3e94938fe25458fabc8f2fc3

Fixes #86
commit c41a2015 (2016-05-22): Add #if for 3.14 kernel change (#87)

Fixes compiling after changes in https://github.com/torvalds/linux/commit/f663dd9aaf9ed124f25f0f8452edf238f087ad50 and https://github.com/torvalds/linux/commit/99932d4fc03a13bb3e94938fe25458fabc8f2fc3

Fixes #86
closed 2017-10-14 by pvaret