pvaret/rtl8192cu-fixes #89: Joint efforts with Raspberry Pi

fran6co, 2016-05-28:
Hi!

   I have been working on this driver for getting cfg80211 support and some fixes. This repository seems to be one of popular ones for supporting 8192cu (there is another one in https://github.com/lwfinger/rtl8192cu).
   Raspberry Pi uses this driver a lot as its most popular dongles use 8192cu. Would be nice to merge this repo with Raspberry Pi and enable all the changes for everyone that it's not using the Raspberry Pi kernel. I suggested this here => https://github.com/raspberrypi/linux/pull/1468#issuecomment-221341071 still waiting for some feedback. One idea would be use this repository or another one supported by Raspberry that gets all the patches + dkms support.
  Here are my PRs:

- https://github.com/raspberrypi/linux/pull/1489
- https://github.com/raspberrypi/linux/pull/1488

dlech, 2016-05-29:
I think it would be better if you put your effort into getting better performance from the new mainline kernel driver. It works, but the main complaint and the reason why people keep falling back to these old drivers is that you don't get the same throughput with the new mainline driver.

fran6co, 2016-05-29:
I don't know enough of driver programming to actually do that, I find it a lot easier to just fix and enable code that's already there.

closed 2017-10-14 by pvaret