pvaret/rtl8192cu-fixes #91: Compilation error against 3.14

paralin, 2016-06-06:
```
  CC [M]  ./rtl8192cu-7329ea8b80948f361f9b4bccc7ae49fa13a6b66f/./core/rtw_br_ext.o
../rtl8192cu-7329ea8b80948f361f9b4bccc7ae49fa13a6b66f/core/rtw_br_ext.c: In function 'nat25_db_handle':
../rtl8192cu-7329ea8b80948f361f9b4bccc7ae49fa13a6b66f/core/rtw_br_ext.c:1449:28: error: implicit declaration of function 'csum_ipv6_magic' [-Werror=implicit-function-declaration]
         hdr->icmp6_cksum = csum_ipv6_magic(&iph->saddr, &iph->daddr,

                            ^
```

Compilation fails here. I tried adding a reference to net/ip6_checksum.h but that resulted in a TON of duplicate declaration errors.

paralin, 2016-06-06:
Note this is against aarch64 (arm 64). It's quite likely that _HAVE_ARCH_IPV6_CSUM is not set on this platform.

eddylai, 2016-11-13:
got the same error message on kernel 3.10 , any solution to fix this?

pvaret, 2016-11-18:
Took a quick look; it indeed doesn't look like that architecture has support for that function. Sorry. :/ Realtek hasn't released a new driver for the 8192cu chipset in years, so it's unlikely they'll ever support this more recent arch. You may want to try the mainline rtl8xxxu driver, or ArchLinux's dkms-8192cu.

closed 2016-11-18 by pvaret