pvaret/rtl8192cu-fixes #94: Cannot install on debian testing kernel 4.6.0-1-amd64

arnoFleming, 2016-06-23:
I've updated my machine to kernel 4.6.0-1-amd64 on debian testing. 

The build fails: 
```bash
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.6.0-1-amd64 -C /lib/modules/4.6.0-1-amd64/build M=/var/lib/dkms/8192cu/1.10/build.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.6.0-1-amd64 (x86_64)
Consult /var/lib/dkms/8192cu/1.10/build/make.log for more information.
```

I've added the make.log, although I'm too much of an end user to understand what it is saying :-o

[make.log.txt](https://github.com/pvaret/rtl8192cu-fixes/files/330830/make.log.txt)

How can I assist to make this driver compile on this kernel?

dlech, 2016-06-23:
This was fixed in #87. Are you sure you have the latest code from git?

arnoFleming, 2016-06-24:
So silly, I tried building the current master branch after I tried building the outdated master. That didn't work. After a reboot I could build it.

closed 2016-06-24 by arnoFleming