pvaret/rtl8192cu-fixes #98: not working with ARM device

fujita-nozomi, 2016-07-15:
when i run 
`sudo dkms install 8192cu/1.10`
it's tell me
`Error! Your kernel headers for kernel 3.4.39 cannot be found.
Please install the linux-headers-3.4.39 package,
or use the --kernelsourcedir option to tell DKMS where it's located
`

dlech, 2016-07-15:
Did you install the `linux-headers-3.4.39` package as the error message told you to?

fujita-nozomi, 2016-07-16:
there's no have linux-headers-3.4.39 and `apt-get` cannot find this package...

dlech, 2016-07-16:
This is a rather old kernel version. What sort of device is this on?

fujita-nozomi, 2016-07-20:
my ARM device is OrangePI PC(Ubuntu system was used at the time，Now is opensuse)
i don't know what's going of this device
look's like "yes i update the kernel to 4.0
but when i reboot the device and `uname -a`it's still say 3.4.39"

roosemberth, 2016-11-24:
Do you have a custom kernel?
what does `uname -r` says?
does the directory /lib/modules/`uname -r` exists?

fujita-nozomi, 2016-12-21:
@roosemberth 
uname -r output this
 `3.4.39`

pvaret, 2017-10-14:
For a kernel this old you almost certainly want to use the driver straight from Realtek's site. There is no point in using this repository, since it's mostly a port of Realtek's driver to recent kernels.

closed 2017-10-14 by pvaret