pvaret/rtl8192cu-fixes #99: (LM18 mate/Ubuntu 16.04) Syslog complaining about not getting signal strength every 6 seconds...

mr-blobbyyy, 2016-08-25:
`(wlan0): error getting signal strength: No such device`

I solved it for now by doing a `sudo systemctl restart network-manager.service`...assuming this will have to be done on every reboot? Googling around...I found [this](https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1594814/comments/5) but that's not applicable in my config file as there is no `dhcp=` entry in `/etc/NetworkManager/NetworkManager.conf`...just the following under "[main]": ```
plugins=ifupdown,keyfile,ofono
dns=dnsmasq
```

mr-blobbyyy, 2016-12-23:
In case anyone is following this, you can automate restarting the NM service (without having to enter the pw each time) by following this guide: http://www.linuxquestions.org/questions/linux-newbie-8/ubuntu-16-04-can-i-delay-starting-the-network-manager-service-4175595871/#post5645098 thankfully I've been lucky these patched drivers still work after all the Ubuntu kernels released within the last few months.

UnknownProgrammer, 2017-01-07:
In december version [1.2.4 of the network-manager package](https://launchpad.net/ubuntu/+source/network-manager/1.2.4-0ubuntu0.16.04.1) was published as proposed for Ubuntu 16.04 (xenial). This version contains the available [fix](https://bugzilla.gnome.org/show_bug.cgi?id=768433). If you don't want to wait until it's released as update for everyone, you could enable the xenial-proposed source, install the package and disable it again.

mr-blobbyyy, 2017-01-07:
@UnknownProgrammer thanks for the suggestion but what about all the other build files on that NM page -- including the g-object introspection data and library files for NM? For x64, wouldn't you need to install [all these "built" debs](https://launchpad.net/ubuntu/+source/network-manager/1.2.4-0ubuntu0.16.04.1/+build/11772934) for the NM build?

Enabling the proposed repo on Mint isn't exactly intuitive either, probably for good reason. They have their own testing repo called "Romeo" you can enable...but you can also add additional repos as well...something like `deb http://archive.ubuntu.com/ubuntu xenial-proposed main restricted universe multiverse` might work and then force version NM?

UnknownProgrammer, 2017-01-08:
@mr-blobbyyy Yes, it is better to update the binary packages, built by the network-manager package, to version number 1.2.4 as well. Replacing only the network-manager package itself basically works, but it could have negative side effects. The files with dev and dbggsym suffix are not necessary.

Sorry, I wrote my post with ubuntu 16.04 in mind. It looks like the packages are not included in Romeo. Adding the repository you mentioned, forcing the version of the packages to 1.2.4 and removing the repository afterwards is the way to go. I did a short test in a VM with LM 18 (Sarah) Mate and it seems to work.

Not all packages from the proposed repository are finally released as update, some are dropped while testing, because they are causing problems. Be careful, if a working system is indispensable for you, you should stay with your method. Otherwise you could end up without any network connection.

Anyhow, the package including the fix is in testing, that’s good news.

UnknownProgrammer, 2017-03-08:
Took a while, but today the network-manager package 1.2.6 was released in the „updates“ pocket. This solves the issue for standard ubuntu 16.04 installations.

mr-blobbyyy, 2017-03-08:
@UnknownProgrammer I've spent the last 4 hours trying to figure out why VPN connections are 404'ing on my mint box. I've about had it with Canonical and ready to install Arch on every machine from now on.

closed 2017-10-14 by pvaret