pvaret/rtl8192cu-fixes #101: Still no solution for TL-WN8200ND?

berndtorino, 2016-10-15:
I am using the TL-WN8200ND and I still could'nt find a proper way to set it up on Ubuntu which I am currently using in version 14.10.

Has there been found a way to make it work on Linux yet?

pvaret, 2016-10-15:
Not that I heard of.

That said, I just tried the native rtl8xxxu driver from kernel 4.8, and it FINALLY seems to work fine with my device (Belkin N300). Maybe give it a go? Don't forget to delete /etc/modprobe.d/blacklist-native-rtl8192.conf.

closed 2016-10-15 by pvaret

intialonso, 2017-04-24:
I know this is closed but wanted to report that to this day this card is still not working. Detect networks but cannot connect, just ask for password again and again.

pvaret, 2017-04-24:
Thanks for the update. The WN8200ND is know not to work. Reportedly it doesn't even work with the official rtl8xxxu driver from the mainline kernel, that I mentioned earlier in this thread. Sorry!

Spenhouet, 2019-12-30:
Is there still no solution?

CGarces, 2019-12-30:
As announced in the README.md, this repository not is maintained, consider open a bug report on your linux distribution in order to try to fix the problem on the mainline code.

Spenhouet, 2019-12-30:
@CGarces I didn't mean a solution from this repository. I did just hope that anyone who followed this issue knows a working workaround for this.

pvaret, 2019-12-31:
For what it's worth, I myself use the driver from the mainline kernel these
days, and it seems to work much more reliably than it used to (kernel 5.3
on the later Ubuntu). It's a long shot, but perhaps they also improved the
support for the TL-WN8200ND?

Depending on which version of that hardware you're using, possibly
https://github.com/luckynzm/tlwn8200nddriver will be of help too. Some
versions of the TL-WN8200ND apparently don't even use the 8192CU chipset.
Sigh.

At any rate, the solution to your problem is not in this repo.

Spenhouet, 2019-12-31:
@pvaret Thanks for your reply.

I'm currently hoping for help from here: https://askubuntu.com/questions/1199677/netplan-wont-connect-to-wireless-network

I am able to scan for networks with the official driver (not sure if that qualifies as "the driver is working correct"). But when trying to connect, it always times out.

I will take a look at the other repo but its for v2.0 of the adapter (I'm using v1.0).