pvaret/rtl8192cu-fixes #102: Adapter was working at 150 Mb/s now it is locked at 72 Mb/s

paralaks, 2016-10-18:
Hello,
I got an Edimax 7811-UN wifi adapter a few days ago and I was having issues with adaptor not waking up from sleep. Edimax support told me to use the fix here https://edimax.freshdesk.com/support/solutions/articles/14000035492-how-to-resolve-ew-7811un-built-in-driver-issues-in-linux-kernel-v3-10-or-higher which is based on source code from this repository.

I applied the solution. Everything works fine except speed is locked to 72 Mb/s which is supposed to be 150 Mb/s. Not sure if this can be solved with a configuration change or there is a bug but I am opening this ticket just in case.
Thanks.

paralaks, 2016-10-18:
It happened to be a configuration setting in the router which was not applied. Rebooting the router with correct settings solved it.

closed 2016-10-18 by paralaks

pvaret, 2016-10-22:
Holy crap, Edimax points people at me to support their hardware? They've got some nerve. Anyway, glad to hear your problem is gone.

As a heads up, it appears that the driver again stopped compiling against recent kernels. But at least the in-kernel driver seems to finally work fine these days, so hopefully this repo will not be needed anymore.

paralaks, 2016-10-22:
I thought you were working for them lol!

They pointed me here: https://edimax.freshdesk.com/support/solutions/articles/14000035492-how-to-resolve-ew-7811un-built-in-driver-issues-in-linux-kernel-v3-10-or-higher 

I think it is reasonable. Linux is open source so I don't see any issues with a driver problem which community can solve and they help us to find it.