pvaret/rtl8192cu-fixes #103: [talk] rtl8xxxu - true love for cheap USB WiFi dongles

dlech, 2016-10-31:
Not an issue, just a point of interest about mainline kernel support for these devices...

There is going to be [a talk about rtl8xxxu at the Linux Plumbers Conference](https://www.linuxplumbersconf.org/2016/ocw/proposals/4089) in a couple days. Hopefully there will be a video and/or a writeup from this. It should be informative.

pvaret, 2016-11-05:
Thanks for the pointer! I emailed Jes to ask about videos and/or slides. :)

strasharo, 2016-11-05:
Awesome!

pvaret, 2016-11-19:
Jes kindly responded! There is no video, but there are slides and [they're available online](https://www.linuxplumbersconf.org/2016/ocw//system/presentations/4089/original/2016-11-02-rtl8xxxu-presentation.pdf).

closed 2016-11-19 by pvaret