pvaret/rtl8192cu-fixes #104: ubuntu 16.04 ASUSTek Nano 802.11n wireless adapter

redheli, 2016-11-05:
Hi,

  recently I install dual system ubuntu 16.04 and 14.04 . the driver works perfect on ubuntu 14.04 with asus nano wifi adapter.

  but no luck on 16.04.
  I followed install steps, no error popup.

I tried [ASUSTek N10 Nano #100](https://github.com/pvaret/rtl8192cu-fixes/issues/100), but still not work.

please help!!!

  here is my system info:
uname -a

> Linux max-Alienware-15 4.4.0-45-generic #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

 lsusb

> Bus 002 Device 002: ID 8087:8000 Intel Corp. 
> Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 001 Device 002: ID 8087:8008 Intel Corp. 
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 003 Device 007: ID 0cf3:3004 Atheros Communications, Inc. AR3012 Bluetooth 4.0
> Bus 003 Device 009: ID 0b05:17ba ASUSTek Computer, Inc. N10 Nano 802.11n Network Adapter [Realtek RTL8192CU]
> Bus 003 Device 003: ID 187c:0528 Alienware Corporation 
> Bus 003 Device 006: ID 046d:c07d Logitech, Inc. 
> Bus 003 Device 005: ID 1bcf:28b6 Sunplus Innovation Technology Inc. 
> Bus 003 Device 004: ID 04f3:2061 Elan Microelectronics Corp. 
> Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> 

sudo lshw -C network
  

> *-network               
>        description: Ethernet interface
>        product: Killer E220x Gigabit Ethernet Controller
>        vendor: Qualcomm Atheros
>        physical id: 0
>        bus info: pci@0000:02:00.0
>        logical name: enp2s0
>        version: 10
>        serial: 34:e6:d7:3c:39:90
>        size: 1Gbit/s
>        capacity: 1Gbit/s
>        width: 64 bits
>        clock: 33MHz
>        capabilities: pm pciexpress msi msix bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
>        configuration: autonegotiation=on broadcast=yes driver=alx duplex=full ip=192.168.1.2 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
>        resources: irq:42 memory:f7400000-f743ffff ioport:d000(size=128)
>   *-network DISABLED
>        description: Wireless interface
>        product: QCA6174 802.11ac Wireless Network Adapter
>        vendor: Qualcomm Atheros
>        physical id: 0
>        bus info: pci@0000:03:00.0
>        logical name: wlp3s0
>        version: 20
>        serial: ac:d1:b8:73:fc:47
>        width: 64 bits
>        clock: 33MHz
>        capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
>        configuration: broadcast=yes driver=ath10k_pci driverversion=4.4.0-45-generic firmware=SW_RM.1.1.1-00157-QCARMSWPZ-1 latency=0 link=no multicast=yes wireless=IEEE 802.11abgn
>        resources: irq:34 memory:f6800000-f69fffff
>   *-network DISABLED
>        description: Wireless interface
>        physical id: 2
>        bus info: usb@3:5
>        logical name: wlx1c872cc63ee0
[... 124 lines cut ...]
> aes_x86_64             20480  1 aesni_intel
> drm_kms_helper        155648  2 i915,nvidia_drm
> lrw                    16384  1 aesni_intel
> gf128mul               16384  1 lrw
> syscopyarea            16384  1 drm_kms_helper
> glue_helper            16384  1 aesni_intel
> sysfillrect            16384  1 drm_kms_helper
> ablk_helper            16384  1 aesni_intel
> sysimgblt              16384  1 drm_kms_helper
> ahci                   36864  3
> cryptd                 20480  3 aesni_intel,ablk_helper
> fb_sys_fops            16384  1 drm_kms_helper
> alx                    36864  0
> nvidia              11931648  71 nvidia_modeset,nvidia_uvm
> libahci                32768  1 ahci
> rtsx_pci               53248  2 rtsx_pci_ms,rtsx_pci_sdmmc
> drm                   364544  9 i915,drm_kms_helper,nvidia_drm
> mdio                   16384  1 alx
> sdhci_acpi             16384  0
> sdhci                  45056  1 sdhci_acpi
> video                  40960  2 i915,dell_wmi
> i2c_hid                20480  0
> hid                   118784  4 i2c_hid,hid_multitouch,hid_generic,usbhid
> fjes                   28672  0
>

pvaret, 2016-11-05:
I'm not sure why your device is shown as disabled. Are "Enable Networking" and "Enable WiFi" both checked in Network Manager?

Note that your other WiFi device (Qualcomm Atheros) is also disabled, so this doesn't look like an issue with this driver.

redheli, 2016-11-07:
@pvaret yes, I think it is not driver problem. maybe it is ubuntu network manager problem, I installed wicd and it can connect to wifi with no problem. 
thanks a lot. your work is awesome.

closed 2016-11-07 by redheli

pvaret, 2016-11-12:
Hey @redheli, glad your problem is solved! And thank you for your kind words, although as I've said elsewhere, the work isn't really ours; this repo is only a repackaging of the official Realtek driver to make it work on more recent Linux versions, that's all. :)