pvaret/rtl8192cu-fixes #107: Wifi not showing up, 16.04 [TP-Link TL-WN823N]

D4RKONION, 2016-12-03:
Hi there.

I followed the instructions in the readme and everything seemed to install without a hitch. However, there is still no wifi options coming up. Please advise

Thanks for all the work you've done on this project!

dlech, 2016-12-04:
There may be some helpful information in these [existing issues about 16.04](https://github.com/pvaret/rtl8192cu-fixes/search?q=16.04&type=Issues&utf8=%E2%9C%93)

D4RKONION, 2016-12-04:
I had read those alright, didn't see anything matching my own issues. One person mentions enabling wifi in the network manager, but I'm seeing no option like that.

samohovets, 2017-02-10:
Same problem

jeremyb31, 2017-02-17:
Some of the TL-WN823N's do not use the rtl8192cu chipset.  If lsusb results show ID 2357:0107, 2357:0108, or 2357:0109 you should use https://github.com/Mange/rtl8192eu-linux-driver as these are rtl8192eu chipsets

robertholding, 2017-04-02:
@jeremyb31 ，this is a right way to install the driver, thank you,

shashishailaj, 2017-04-09:
@robertholding  , It  (https://github.com/Mange/rtl8192eu-linux-driver ) worked for me without a problem with TP-LINK TL-WN823N V2 WiFi adapter running on KDE neon 4.9.2 (based on Ubuntu 16.04 LTS) Kernel 4.8. 

Try to use the instructions on the github page https://github.com/Mange/rtl8192eu-linux-driver . It must work.

robertholding, 2017-04-22:
@shashishailaj thanks for your help. it works!

closed 2017-10-14 by pvaret