pvaret/rtl8192cu-fixes #108: Compile Error on power PC

reboutte, 2016-12-28:
hello, 
on imac G4 (a bit old )
debian jessie

kernel  3.16.0-4-powerpc
adapter
 ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter

compil error ( file /var/lib/dkms/8192cu/1.10/build/make.log )

DKMS make.log for 8192cu-1.10 for kernel 3.16.0-4-powerpc (ppc)
mercredi 28 décembre 2016, 22:22:11 (UTC+0100)
make: Entering directory '/usr/src/linux-headers-3.16.0-4-powerpc'
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-4-powerpc'
  LD      /var/lib/dkms/8192cu/1.10/build/built-in.o
  CC [M]  /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o
In file included from /var/lib/dkms/8192cu/1.10/build/include/byteorder/little_endian.h:30:0,
                 from /var/lib/dkms/8192cu/1.10/build/include/rtw_byteorder.h:31,
                 from /var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1233,
                 from /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.c:23:
/var/lib/dkms/8192cu/1.10/build/include/byteorder/swab.h:114:29: error: conflicting types for ‘__fswab16’
 __inline static const __u16 __fswab16(__u16 x)
                             ^
In file included from /usr/src/linux-headers-3.16.0-4-common/include/linux/swab.h:4:0,
                 from /usr/src/linux-headers-3.16.0-4-common/include/uapi/linux/byteorder/big_endian.h:12,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/byteorder/big_endian.h:4,
                 from /usr/src/linux-headers-3.16.0-4-common/arch/powerpc/include/uapi/asm/byteorder.h:13,
                 from /usr/src/linux-headers-3.16.0-4-common/include/asm-generic/bitops/le.h:5,
                 from /usr/src/linux-headers-3.16.0-4-common/arch/powerpc/include/asm/bitops.h:279,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/bitops.h:36,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/kernel.h:10,
                 from /usr/src/linux-headers-3.16.0-4-common/include/asm-generic/bug.h:13,
                 from /usr/src/linux-headers-3.16.0-4-common/arch/powerpc/include/asm/bug.h:127,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/bug.h:4,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/thread_info.h:11,
                 from /usr/src/linux-headers-3.16.0-4-common/include/asm-generic/preempt.h:4,
                 from arch/powerpc/include/generated/asm/preempt.h:1,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/preempt.h:18,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/spinlock.h:50,
                 from /var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:754,
                 from /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.c:23:
/usr/src/linux-headers-3.16.0-4-common/include/uapi/linux/swab.h:46:41: note: previous definition of ‘__fswab16’ was here
 static inline __attribute_const__ __u16 __fswab16(__u16 val)
                                         ^
In file included from /var/lib/dkms/8192cu/1.10/build/include/byteorder/little_endian.h:30:0,
                 from /var/lib/dkms/8192cu/1.10/build/include/rtw_byteorder.h:31,
                 from /var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1233,
                 from /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.c:23:
/var/lib/dkms/8192cu/1.10/build/include/byteorder/swab.h:122:29: error: conflicting types for ‘__fswab32’
 __inline static const __u32 __fswab32(__u32 x)
                             ^
In file included from /usr/src/linux-headers-3.16.0-4-common/include/linux/swab.h:4:0,
                 from /usr/src/linux-headers-3.16.0-4-common/include/uapi/linux/byteorder/big_endian.h:12,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/byteorder/big_endian.h:4,
                 from /usr/src/linux-headers-3.16.0-4-common/arch/powerpc/include/uapi/asm/byteorder.h:13,
                 from /usr/src/linux-headers-3.16.0-4-common/include/asm-generic/bitops/le.h:5,
                 from /usr/src/linux-headers-3.16.0-4-common/arch/powerpc/include/asm/bitops.h:279,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/bitops.h:36,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/kernel.h:10,
                 from /usr/src/linux-headers-3.16.0-4-common/include/asm-generic/bug.h:13,
                 from /usr/src/linux-headers-3.16.0-4-common/arch/powerpc/include/asm/bug.h:127,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/bug.h:4,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/thread_info.h:11,
                 from /usr/src/linux-headers-3.16.0-4-common/include/asm-generic/preempt.h:4,
                 from arch/powerpc/include/generated/asm/preempt.h:1,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/preempt.h:18,
                 from /usr/src/linux-headers-3.16.0-4-common/include/linux/spinlock.h:50,
                 from /var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:754,
                 from /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.c:23:
/usr/src/linux-headers-3.16.0-4-common/include/uapi/linux/swab.h:57:41: note: previous definition of ‘__fswab32’ was here
 static inline __attribute_const__ __u32 __fswab32(__u32 val)
                                         ^
/usr/src/linux-headers-3.16.0-4-common/scripts/Makefile.build:262: recipe for target '/var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o' failed
make[3]: *** [/var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o] Error 1
/usr/src/linux-headers-3.16.0-4-common/Makefile:1352: recipe for target '_module_/var/lib/dkms/8192cu/1.10/build' failed
make[2]: *** [_module_/var/lib/dkms/8192cu/1.10/build] Error 2
Makefile:181: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/usr/src/linux-headers-3.16.0-4-powerpc'

any idea ??

dlech, 2016-12-28:
Remove the conflicting definitions from `include/byteorder/swab.h` in this repository?

reboutte, 2016-12-29:
how can i do that ??

dlech, 2016-12-29:
Edit the file in the directory that was created were checked out when you did `git clone`. Then run the `dkms` commands again.

reboutte, 2016-12-29:
but i don't know how to remove the conflicting definitions ...
edit a file, no problem, but , i don't "speak C++  fluently"

reboutte, 2016-12-29:
Hello, 
the problem come from the ppc kernel, it work perfectly with intel kernel....
any idea?

dlech, 2016-12-29:
You could try replacing

    #include <bytecodes/swab.h>

with

    #include <linux/swab.h>

in [include/byteorder/little_endian.h](https://github.com/pvaret/rtl8192cu-fixes/blob/6a58e2f77d75ca9a3b80868a344ed4e2ea1816df/include/byteorder/little_endian.h#L30)

closed 2017-10-14 by pvaret

vareille, 2021-04-17:
Hi, same here, on a old mac g3 ppc running debian jessie.
I've did replace  #include <byteorder/swab.h> with#include <linux/swab.h>

then I get the following. any idea ?

Arch ppc is not supported with CONFIG_FTRACE_MCOUNT_RECORD at /usr/src/linux-headers-3.16.0-6-common/scripts/recordmcount.pl line 378.
/usr/src/linux-headers-3.16.0-6-common/scripts/Makefile.build:262: recipe for target '/home/officer/rtl8192cu-fixes/core/rtw_cmd.o' failed