pvaret/rtl8192cu-fixes #110: Not working any longer on F25, kernel 4.9

GregoryGoncalves, 2017-02-06:
As the title says. I was using fedora 25 about two weeks ago with this fix, now i reinstalled fedora and it hasn't worked at all, it deactivates the wifi completely, tried multiple times with kernels 4.8 and 4.9.
I'm not very experienced so i'm not sure what else i can help with. I can provide any necessary debugging logs if necessary.

duperray, 2017-02-06:
Hi, I'm using debian unstable, compiled ok, loaded 8192cu with modprobe, but I get:
Feb  6 10:02:09 metro kernel: [  516.291725] IPv6: ADDRCONF(NETDEV_UP): wlxe4f4c63a168a: link is not ready
Feb  6 10:02:09 metro NetworkManager[441]: <info>  [1486371729.5310] device (wlxe4f4c63a168a): supplicant interface state: inactive -> disabled

duperray, 2017-02-06:
forgot to say in testing/unstable debian we have 4.9 kernel too

GregoryGoncalves, 2017-02-06:
I've had no problems with debian jessie. I was using up until last week.
I bought another adaptor however, gave up on fixing the issue.

schwella, 2017-02-09:
I've got the same problem under Fedora. Even using one of the earlier kernels didn't work after the update. I cann tell that when reading dmesg I can see the same issues mentioned by @duperray

Pant, 2017-02-17:
I have the same problem. Workaround:
`nmcli con up id <AP_Name>`

duperray, 2017-02-17:
It could not see networks, so I clicked connect to hiden network, connected without problem and then
all network suddendly appeared!
Seems to work now.

gabrielnn77, 2017-04-02:
i tested in F 25, kernel 4.10.6
same problem

gabrielnn77, 2017-04-05:
new try, on f 25, kernel 4.10.8-200.fc25.x86_64
#lsmod | grep 8192
8192cu                536576  0
snd_hda_core           81920  5 snd_hda_intel,snd_hda_codec,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_codec_realtek
r8169                  81920  0

NetworkManager[941]: <info>  [1491356448.4214] device (wlan2): supplicant interface state: scanning -> associating
NetworkManager[941]: <info>  [1491356452.3670] device (wlan2): supplicant interface state: associating -> disconnected
NetworkManager[941]: <info>  [1491356452.4672] device (wlan2): supplicant interface state: disconnected -> scanning
NetworkManager[941]: <warn>  [1491356457.3690] device (wlan2): Activation: (wifi) association took too long
NetworkManager[941]: <info>  [1491356457.3691] device (wlan2): state change: config -> need-auth (reason 'none') [50 60 0]
NetworkManager[941]: <warn>  [1491356457.3712] device (wlan2): Activation: (wifi) asking for new secrets
NetworkManager[941]: <warn>  [1491356457.3934] device (wlan2): No agents were available for this request.
NetworkManager[941]: <info>  [1491356457.3935] device (wlan2): state change: need-auth -> failed (reason 'no-secrets') [60 120 7]
NetworkManager[941]: <info>  [1491356457.3937] manager: NetworkManager state is now CONNECTED_LOCAL
NetworkManager[941]: <warn>  [1491356457.3943] device (wlan2): Activation: failed for connection 'wlan2'
NetworkManager[941]: <info>  [1491356457.3949] device (wlan2): state change: failed -> disconnected (reason 'none') [120 30 0]
kernel: IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready
NetworkManager[941]: <info>  [1491356459.3566] device (wlan2): supplicant interface state: scanning -> inactive

wpa_supplicant.log:
wlan2: Trying to associate with 18:d6:c7:cf:2f:a4 (SSID='TP-LINK' freq=2452 MHz)
wlan2: CTRL-EVENT-DISCONNECTED bssid=18:d6:c7:cf:2f:a4 reason=0
wlan2: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="TP-LINK" auth_failures=1 duration=10 reason=CONN_FAILED
wlan2: CTRL-EVENT-SSID-REENABLED id=0 ssid="TP-LINK"
wlan2: Trying to associate with 18:d6:c7:cf:2f:a4 (SSID='TP-LINK' freq=2452 MHz)
wlan2: CTRL-EVENT-DISCONNECTED bssid=18:d6:c7:cf:2f:a4 reason=0
wlan2: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="TP-LINK" auth_failures=2 duration=20 reason=CONN_FAILED

tomchiverton, 2017-06-01:
Same issues on Ubuntu 16.04 LTS (with or without the HWE updated kernel packages)
4.8.0-53 fails to connect. 

syslog says

Jun  1 10:23:48 ev34 wpa_supplicant[1660]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="XXXXXXXXX"
Jun  1 10:23:48 ev34 wpa_supplicant[1660]: wlan0: Trying to associate with xx:xx:df:xx:xx:xx (SSID='XXXXXXXXX' freq=2462 MHz)
Jun  1 10:23:48 ev34 wpa_supplicant[1660]: wlan0: Association request to the driver failed

sunshinertoo, 2017-08-07:
4.9.30 kernel from stretch-backports on debian stable does not work any longer for the ASUS USB-N13 WiFi Dongle.
The module gets loaded.

thisch, 2017-09-30:
I'm using the rtl8xxxu driver instead, which works fine (kernel 4.9). 
See  https://bugzilla.redhat.com/show_bug.cgi?id=1421385

pvaret, 2017-10-14:
Seconding this: the rtlxxxu works pretty fine these days. You may still have to troubleshoot NetworkManager, though: in particular, consider [disabling power management](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Disabling it made the rtlxxxu driver work well for me.

closed 2017-10-14 by pvaret