pvaret/rtl8192cu-fixes #111: Build fails with kernel 4.11.0-rc1

Ito76, 2017-03-13:
Hello, I'm trying to build the 8192cu module with kernel 4.11.0-rc1 but I get these errors, I guess because of some API change:
http://pastebin.com/raw/gyWTcheE

No errors when building with previous 4.10.x
Thank you in advance

deri82, 2017-03-25:
I have same issue with 4.11 series.

deri82, 2017-04-03:
Does this help?

Cisco: http://lxr.free-electrons.com/ident?i=allow_signal
Cisco: also the not found function is not documented for 4.11
Cisco: get the source for the driver
Cisco: open include/osdep_service.h
Cisco: and put #include <linux/signal.h>
Cisco: under #define __OSDEP_SERVICE_H_
Cisco is now Offline.

Pant, 2017-04-07:
Problem persists in fedora 26-alpha: "4.11.0-0.rc5.git0.1.fc26.x86_64"

pvaret, 2017-04-07:
​Welp. If there is such a thing as a kernel documentation that says "we
broke that API as of version 4.XX.YY, use this new one instead", I failed
to track it down.

deri82: it would help indeed if it covered the version that broke the API
(4.11, apparently), but it doesn't.

But until it does, I don't think it worth anyone's time to make this driver
work on an unreleased kernel. Your time will be better spent helping debug
the mainline kernel for this device range (rtl8xxx).

PR #114 (2017-04-18): Fix build with kernel 4.11
Add compatibility for 4.11
Also patch kernels 4.8 & 4.7

Close #111
commit 93b3b958 (2017-04-18): Fix build with kernel 4.11
Patch for kernels 4.8 & 4.7

Close #111
closed 2017-04-18 by pvaret