pvaret/rtl8192cu-fixes #120: Compilation error with 4.12

maxime-tournier, 2017-09-02:
Here's the log: 
```
DKMS make.log for 8192cu-1.10 for kernel 4.12.0-12-generic (x86_64)
Sat Sep  2 10:26:40 CEST 2017
make: Entering directory '/usr/src/linux-headers-4.12.0-12-generic'
  LD      /var/lib/dkms/8192cu/1.10/build/built-in.o
  CC [M]  /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8192cu/1.10/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8192cu/1.10/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8192cu/1.10/build/core/rtw_io.o
In file included from /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.c:23:0:
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h: In function ‘thread_enter’:
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1482:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
```

natanfelles, 2017-09-18:
Same issue here. On Debian Testing:

```
$ sudo dkms install 8192cu/1.10

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j2 KERNELRELEASE=4.12.0-1-amd64 -C /lib/modules/4.12.0-1-amd64/build M=/var/lib/dkms/8192cu/1.10/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.12.0-1-amd64 (x86_64)
Consult /var/lib/dkms/8192cu/1.10/build/make.log for more information.
```

```
$ cat /var/lib/dkms/8192cu/1.10/build/make.log
DKMS make.log for 8192cu-1.10 for kernel 4.12.0-1-amd64 (x86_64)
seg set 18 18:37:51 -03 2017
make: Entering directory '/usr/src/linux-headers-4.12.0-1-amd64'
  LD      /var/lib/dkms/8192cu/1.10/build/built-in.o
  CC [M]  /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8192cu/1.10/build/core/rtw_security.o
In file included from /var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.c:23:0:
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h: In function ‘thread_enter’:
In file included from /var/lib/dkms/8192cu/1.10/build/core/rtw_security.c:23:0:
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h: In function ‘thread_enter’:
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1482:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1482:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h: In function ‘flush_signals_thread’:
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h: In function ‘flush_signals_thread’:
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1495:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1495:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1497:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
/var/lib/dkms/8192cu/1.10/build/include/osdep_service.h:1497:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
/var/lib/dkms/8192cu/1.10/build/core/rtw_security.c: In function ‘aes_cipher’:
/var/lib/dkms/8192cu/1.10/build/core/rtw_security.c:1504:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/var/lib/dkms/8192cu/1.10/build/core/rtw_security.c:1507:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
/var/lib/dkms/8192cu/1.10/build/core/rtw_security.c: In function ‘aes_decipher’:
/var/lib/dkms/8192cu/1.10/build/core/rtw_security.c:1878:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/var/lib/dkms/8192cu/1.10/build/core/rtw_security.c:1881:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.12.0-1-common/scripts/Makefile.build:307: recipe for target '/var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o' failed
make[3]: *** [/var/lib/dkms/8192cu/1.10/build/core/rtw_cmd.o] Error 1
make[3]: ** Esperando que outros processos terminem.
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.12.0-1-common/scripts/Makefile.build:307: recipe for target '/var/lib/dkms/8192cu/1.10/build/core/rtw_security.o' failed
make[3]: *** [/var/lib/dkms/8192cu/1.10/build/core/rtw_security.o] Error 1
/usr/src/linux-headers-4.12.0-1-common/Makefile:1532: recipe for target '_module_/var/lib/dkms/8192cu/1.10/build' failed
make[2]: *** [_module_/var/lib/dkms/8192cu/1.10/build] Error 2
Makefile:152: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/usr/src/linux-headers-4.12.0-1-amd64'
```

CGarces, 2017-09-22:
I can't reproduce the error.

[Compiling for 4.12.0](https://travis-ci.org/CGarces/rtl8192cu-fixes/jobs/278766133) works fine on travis build.

Kernel 4.12 was fixed on 8757801e16e38e460d0b5ab7f27356e923bafd86 

@pvaret, are you interested in the Travis CI integration with your repository?

natanfelles, 2017-09-23:
Uolll! Hey, @CGarces. Very thanks for your reply and sorry for my mistake.

I was trying to compile the project that I was download at march 26. Then the errors was occurring.

I look that the PR #116 was accepted on may 31.

I downloaded the current project and retried and now it works fine.

Again, very thanks.

This was my last installation:

```
$ uname -a
Linux xxx.xxx.xxx 4.12.0-1-amd64 #1 SMP Debian 4.12.6-1 (2017-08-12) x86_64 GNU/Linux
```

```
$ sudo dkms remove 8192cu/1.10 --all

-------- Uninstall Beginning --------
Module:  8192cu
Version: 1.10
Kernel:  4.9.0-2-amd64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

8192cu.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.9.0-2-amd64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.....

Backing up initrd.img-4.9.0-2-amd64 to /boot/initrd.img-4.9.0-2-amd64.old-dkms
Making new initrd.img-4.9.0-2-amd64
(If next boot fails, revert to initrd.img-4.9.0-2-amd64.old-dkms image)
update-initramfs......

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.10
completely from the DKMS tree.
------------------------------
Done.
```

```
$ sudo dkms add ./rtl8192cu-fixes-master

Creating symlink /var/lib/dkms/8192cu/1.10/source ->
                 /usr/src/8192cu-1.10

DKMS: add completed.
```

```
$ sudo dkms install 8192cu/1.10

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j2 KERNELRELEASE=4.12.0-1-amd64 -C /lib/modules/4.12.0-1-amd64/build M=/var/lib/dkms/8192cu/1.10/build.............
cleaning build area...

DKMS: build completed.

8192cu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.12.0-1-amd64/updates/dkms/

depmod...

Backing up initrd.img-4.12.0-1-amd64 to /boot/initrd.img-4.12.0-1-amd64.old-dkms
Making new initrd.img-4.12.0-1-amd64
(If next boot fails, revert to initrd.img-4.12.0-1-amd64.old-dkms image)
update-initramfs.....

DKMS: install completed.
```

```
$ sudo depmod -a
```

```
$ sudo cp ./rtl8192cu-fixes-master/blacklist-native-rtl8192.conf /etc/modprobe.d/
```

Reboot.

pvaret, 2017-09-23:
@CGarces Hi, that's a good idea actually. Do you have pointers to get started?

closed 2017-09-23 by pvaret