pvaret/rtl8192cu-fixes #123: Kubuntu 16.10, 4.13 kernel:

Vantuz, 2017-11-07:


pvaret, 2017-11-07:
Hi! Thank you for taking the time to write this thorough ticket.

In general, bugs in the driver should be reported to Realtek, as this repo only packages their driver to make it compile on recent kernels. That said, from your report, based on the line that says "GENERAL.DRIVER", it looks like your device is using the rtl8192cu driver that comes with the kernel. This driver should appear as 8291cu, without the 'rtl' prefix. So that's probably your problem.

*That* said, now, the rtl8xxxu driver should work pretty well these days, given a proper power management configuration. That's what I'm currently using, to be honest. :) Probably you'll want to look in that direction as a long-term solution?

closed 2017-11-07 by pvaret

Vantuz, 2017-11-08:
> based on the line that says "GENERAL.DRIVER", it looks like your device is using the rtl8192cu driver that comes with the kernel. 

I don't know why this line says so. In fact, I even tryed to rename `/lib/modules/4.13.0-16-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192cu/rtl8192cu.ko` file, which, I guess, is rtl8192cu which comes with kernell and it had no effect. So, I guess, it is not the module I am currently using.
But if the problem is really that I am using the wrong driver, then, I guess, something is wrong with installation process, which, I hope, is possible to fix.

I tried the rtl8xxxu driver, it neither disconnects from AP nor halts data transfer but works REALLY slow.
Result of speedtest.net test:
[image]
I can't cope with this speed. I get much better performance with 8192cu driver when it works. I did the power management config as described [here](https://askubuntu.com/questions/695867/disable-wifi-power-management/961460#961460). Still, no luck.