pvaret/rtl8192cu-fixes #124: Instruction build Centos7/El7

naatje80, 2017-11-22:
For those struggling with the Installation of the module on REL7/Centos 7: It took me couple of days to figure out why the driver was not working, although the compilation went successfully. 

Looking at the /var/log/messages reviled that NetworkManager does not seem to recognize the driver, although this was working fine in RHEL 6/Centos 6. It seems that NetwokrManager was no longer build with wext support (this is only done when the rpms are build on and fore a fedora system). Changing the line --with-wext=yes, solved the issue for me.

To manually rebuild the rpms:
1. Download the source rpm for NetworkManager, or execute the following steps:
yum -y install yum-utils
yumdownloader --source NetworkManager
2. Install the source rpm: rpm -ivh Network-Manager-1.8.0-<version>.el7_<elversion>.src.rpm
3. yum -y install rpm-build
4. cd $HOME/rpmbuild
5. rpmbuild --define "debug_package %{nil}" --define "%dist .el7_4" -ba SPECS/NetworkManager.spec
6. Force the reinstallation of the rpms: sudo rpm -ivh RPMS/noarch/* RPMS/x86_64/* --replacefiles --replacepkgs
7. Restart the NetworkManager: sudo systemctl restart NetworkManager. Or, if that does not help, reboot the system.
8. Now the driver should also be recognized by the system again. Check the /var/log/messages or journalctl -t NetworkManager to see if the driver is indeed recognized by NetworkManager

@pvaret: Would you like me to create a pull request to add these instructions? Of course I would rewire it in a more formal form....

naatje80, 2017-11-22:
Forgot one line: edit: SPECS/NetworkManager.spec and change --with-wext=no to --with-wext=yes before building the rpms (between step 4 and 5)

pvaret, 2018-02-02:
Hi,

Yes, it would be nice to have these instructions. Can you put them in a new file, say, README.centos? I'll link it in the main README.

saltylamon, 2018-08-09:
build error on 3.10

ake -j2 KERNELRELEASE=3.10.0-862.el7.x86_64 -C /lib/modules/3.10.0-862.el7.x86_64/build M=/var/lib/dkms/8192cu/1.11/build..........(bad exit status: 2)
Error! Bad return status for module build on kernel: 3.10.0-862.el7.x86_64 (x86_64)
Consult /var/lib/dkms/8192cu/1.11/build/make.log for more information.
[root@cortex code]# tail /var/lib/dkms/8192cu/1.11/build/make.log
  ^
/var/lib/dkms/8192cu/1.11/build/os_dep/linux/ioctl_cfg80211.c:5275:2: warning: (near initialization for ‘rtw_cfg80211_ops.del_station’) [enabled by default]
/var/lib/dkms/8192cu/1.11/build/os_dep/linux/ioctl_cfg80211.c:5292:2: warning: initialization from incompatible pointer type [enabled by default]
  .mgmt_tx = cfg80211_rtw_mgmt_tx,
  ^
/var/lib/dkms/8192cu/1.11/build/os_dep/linux/ioctl_cfg80211.c:5292:2: warning: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx’) [enabled by default]
make[1]: *** [/var/lib/dkms/8192cu/1.11/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [_module_/var/lib/dkms/8192cu/1.11/build] Error 2
make: Leaving directory `/usr/src/kernels/3.10.0-862.el7.x86_64'

pvaret, 2018-11-07:
@naatje80: all right, I'll be closing this, given the long silence. Please reopen or send a PR with your suggested instructions, as needed.

@saltylamon: so, three things: this ticket is about something else and not the right place for this comment; your log excerpt shows warning but not the actual error; and that kernel is really ancient and who even knows what it's even doing. :) It may be that a recent patch to make this repo compile against a newer kernel broke something for kernel 3.10. At this point I'm not very keen on fixing that. Please consider upgrading to a kernel newer than 4.4 and using the native rtl8xxxu driver.

closed 2018-11-07 by pvaret