pvaret/rtl8192cu-fixes #125: adapt to new /proc

rwst, 2017-11-29:
As you mention in Current Status the driver was not adapted to newer kernels, and this leads to complete silence in the kernel log (together with working wlan at least some sign that the driver replaced the kernel drivers and is working). But better would be to have some logs because net services may rely on /proc telling them things.

So I think this is an issue to fix (unfortunately I'm not into hardware).

pvaret, 2017-11-29:
The driver does write kernel logs. The /proc issue is unrelated to logs.

If you're not seeing the driver's logs in dmesg, then it probably means
it's just not getting loaded. Make sure you've correctly blacklisted the
default kernel driver as indicated in the installation instructions.

Better yet, try switching to the rtl8xxxu driver from the mainline kernel.
It works pretty fine these days.

rwst, 2017-11-30:
Thanks for the info, esp. about rtl8xxxu. I'd suggest putting it in the README.md right at the top. Also blacklisting on OpenSuSE now only works via the command `blacklist`, and I succeeded loading rtl8xxxu by giving this in the conf file:
```
blacklist rtl8192cu
blacklist rtl8192c_common
```
dmesg has now:
```
[    8.740718] usb 1-2: rtl8192cu_parse_efuse: dumping efuse (0x80 bytes):
[    8.740790] usb 1-2: rtl8xxxu: Loading firmware rtlwifi/rtl8192cufw_TMSC.bin
[    9.152134] usbcore: registered new interface driver rtl8xxxu
[   10.258472] usb 1-2: rtl8xxxu_bss_info_changed: HT supported
[   30.359010] usb 1-2: rtl8xxxu_ampdu_action: IEEE80211_AMPDU_RX_START
```
Too early to say it's an improvement.
People might also need the latest firmware file `rtl8192cufw_TMSC.bin`.

pvaret, 2017-12-08:
Good idea, thanks. Done!

closed 2017-12-08 by pvaret