pvaret/rtl8192cu-fixes #128: moniter mode phy is null

mohammedhassannai, 2017-12-22:
after i downloaded your fix on kali linux when i try to start moniter nothing happened 
airmon-ng says that phy is null

pvaret, 2017-12-23:
This driver is obsolete. You almost certainly want to use the rtl8xxxu driver that comes with recent kernels. It works pretty well these days. You may have to blacklist the old rtl8192cu driver that some distros still load by default, though.

closed 2017-12-23 by pvaret

mohammedhassannai, 2017-12-23:
I am running kali with kernel  SMP Debian 4.14.2-1kali1 which rtl81xxxu driver wont run on it as it run on 4.4 and newer and i use the blacklist conf in your fix and still the phy is null plz help me i can not start moniter mode on my device and i have much work to do

pvaret, 2017-12-23:
Err. 4.14.2 is newer than 4.4, isn't it...?